etchdroid/etchdroid #5: Ability to create Windows USB installers

Depau, 2018-09-10:
Relevant information: https://rys.pw/operating_systems#windows_images

### Requirements:

- ISO9660
  - [ ] Mount Windows ISOs (possibly without root, maybe without actually mounting them but reading the image directly with some library)
- FAT32
  - [ ] Create FAT32 filesystems (shouldn't be too hard)
  - [x] Read/write FAT32 filesystems (libaums can do that)
- NTFS (needs investigation on what we can do on Android with NTFS-3G)
  - [ ] Create NTFS filesystems
  - [ ] Read/write NTFS filesystems
- Bootloader options
  - UEFI (needs discussion with F-Droid maintainers on how to deal with Secure Boot signed images)
    - [x] rEFInd (very easy to use, build, install and configure)
    - [x] systemd-boot (same as above)
    - [ ] GRUB (easy to use and configure, needs investigation on how `grub-install` can be built for Android while still building the bootloader for x86)
  - BIOS (needs investigation on how the boot sector can be created, I'm pretty sure a quick and dirty hack can be found)
    - [ ] GRUB
    - [ ] SysLinux (can it boot Windows)
    - [ ] Others

Once the requirements are met, the app will have to:
- Create a MBR partition table on the USB drive
- Add two partitions, one formatted with FAT32 (EFI System Partition) and the other with NTFS
- Copy the image *content* (not the image itself) to the NTFS
- Install a (signed) UEFI bootloader to the ESP
- Install a BIOS bootloader

Gen0, 2020-06-04:
Very confused as this seems mucho complicated. Understanding the source and destination partitions should only be relevant when writing to pre-partitioned drives AFAIK. Block for Block Iso9660 is far more reliable and the format is extensible to ISO 13490 for read write access. Even Windows itself will not format the drive as FAT32, but may choose to retain a FAT32 partition in the first 250MB for EFI System Reserved partition. Due to the likelihood of a BIOS not natively supporting either ISO13490 or NTFS, ISO9660 is the best choose for a one click no options app like you have.

Depau, 2020-06-04:
I'm not sure what you're referring to.

Whatever filesystem can be used, as long as it's supported by the Windows installer **on USB drives** (for MBR) and by most firmwares + Windows (UEFI).

Windows does support FAT32, NTFS, but also ISO9660 and UDF since it's also able to boot the installer from CDs and DVDs. However:

- Dumping the ISO to a USB drive doesn't work, but I'm not surprised
- Maybe creating a GPT/MBR partition table on the drive and then copying bit-by-bit the UDF filesystem inside a partition will work? I've never tested or investigated this, I'm thinking of it only now.

Fat32 is probably the easiest choice since 1) it's already implemented in libaums, 2) tackling the 4GB issue isn't too hard with `wimlib` and 3) it's supported by all UEFI firmwares (including Apple devices).

Going with ISO9660 would add it up to the filesystems that need to be added to libaums (on top of UDF).

If you're referring to the fact that I wrote ISO9660 in the tasklist in the first post, that's an error, Windows ISOs use UDF if I remember correctly. I'll correct it.

Either way I'm quite busy right now and EtchDroid needs some major clean up before this is implemented. I'm not happy at all with the way it's currently implemented and I'd like to make it more "pluggable", since I'm planning to distribute the Windows ISO functionality as a separate paid app plugin on the Play Store (still free on F-Droid and open-source GPLv3-licensed on GitHub for whoever wants to build it themselves — Termux plugins-style).

Tilloboy1984, 2026-09-08:
Salom

Tilloboy1984, 2026-09-08:
Windows 10 64-bit iso