etchdroid/etchdroid #6: Ability to create macOS installers

Depau, 2018-09-10:
Relevant information:
- https://support.apple.com/en-us/HT201372
- https://www.insanelymac.com/forum/topic/329828-making-a-bootable-high-sierra-usb-installer-entirely-from-scratch-in-windows-or-linux-mint-without-access-to-mac-or-app-store-installerapp/

As of now this feature requires a lot of investigation (i.e. I would at least need to be able to follow the InsanelyMac guide and adapt it to use free software only)

theoamaya, 2019-03-24:
Thanks man your the best

benthetechguy, 2022-06-04:
> * [ ]  Ability to mount HFS+ **read-write**
>   
>   * I doubt most phones will have the `hfsplus` kernel module; a FUSE implementation must be found

https://github.com/0x09/hfsfuse could be helpful

closed 2025-05-02 by depau