etchdroid/etchdroid #7: No USB Drives Detected.

ifailedprek, 2018-09-27:
I'm trying to repair my Windows 10 using the ISO and I don't exactly have a pendrive that connects to my phone, but I do have a charger. Everything is connected but after I choose the ISO file it says no USB drives detected. Does this require a pendrive that connects to my phone then remove and connect to my computer? I have a Galaxy S9+. I have a separate normal pendrive but I can't get access to a computer and all I have is my phone.

Depau, 2018-09-27:
Hi, of course you need a USB drive. This app writes images to USB drives.

Also it does not support Windows 10 images yet, and though it's planned for a future release it will very likely require root.

I'm closing this as it's not an issue.

closed 2018-09-27 by Depau

geribabldi, 2018-12-18:
I have a sandisc usb stick, your app isnt able to detect it. Is it maybe a format issue? I used the standard android format

Depau, 2018-12-19:
No, it's not a format issue. The app doesn't care about the partition layout as it's going to erase it anyway at some point during the process.

Make sure your phone supports USB OTG. If USB drives are not detected it means that Android did not detect it either. That can happen for a number of reasons, such as the phone not supporting USB OTG, the USB adapter being broken, the USB drive being broken, and others.