etchdroid/etchdroid #9: Unfortunately, EtchDroid has stopped.

omkanLoc, 2018-10-15:
Hello,
I have installed and used EtchDroid on a Lenovo TB-X304F (Android version 7.1.1) and it works great.  I want to install it on 4 other tablets, all are Lenovo TB-X103F (Android version 6.0.1).  I am using the same OTG usb reader that works on the TB-X304F, but it will not work on any of the TB-X103F tablets.  These tablets support OTG.  The sd card and reader is recognized when I plug it into the TB-X103F.  EtchDroid installed without error.  The app opens ok, but when I click "Write raw image or ISO" the app closes and gives me "Unfortunately, EtchDroid has stopped." error.  The tablet software has been updated to latest version.  On the TB-X103F tablets I did notice in the play store that there was a "flag/notice" under the Install button that says "This app may not be optimized for your device"  I do not know it this has anything to do with the app closing unexpectedly.  Or if there is another tablet related incompatibility.  Any help would be greatly appreciated!!

Depau, 2018-10-16:
Please provide the exception stack trace. If the app crashes as soon as you click a button on the first screen it means your phone doesn't provide the storage access framework. If play store says you can't install the app it means it won't work on it, disabling my app from running on random devices is not among my hobbies.

If the app crashes when finally writing the image after having selected it, please provide the "write failed" screen error message.

omkanLoc, 2018-10-16:
Hello Davide,
Thanks for your response and help. The app crashes as soon as I click a button on the first screen, so I guess it is a SAF issue, but I think my Android version should be compatible in this sense.
Here is a stack trace:
--------- beginning of system
--------- beginning of main
--------- beginning of crash
10-16 15:44:57.511 11144 11144 E AndroidRuntime: FATAL EXCEPTION: main
10-16 15:44:57.511 11144 11144 E AndroidRuntime: Process: eu.depau.etchdroid, PID: 11144
10-16 15:44:57.511 11144 11144 E AndroidRuntime: java.lang.IllegalArgumentException: Expected receiver of type eu.depau.etchdroid.activities.StartActivity, but got android.view.ContextT
hemeWrapper
10-16 15:44:57.511 11144 11144 E AndroidRuntime: at java.lang.reflect.Method.invoke(Native Method)
10-16 15:44:57.511 11144 11144 E AndroidRuntime: at android.view.View$DeclaredOnClickListener.onClick(View.java:4454)
10-16 15:44:57.511 11144 11144 E AndroidRuntime: at android.view.View.performClick(View.java:5205)
10-16 15:44:57.511 11144 11144 E AndroidRuntime: at android.view.View$PerformClick.run(View.java:21164)
10-16 15:44:57.511 11144 11144 E AndroidRuntime: at android.os.Handler.handleCallback(Handler.java:739)
10-16 15:44:57.511 11144 11144 E AndroidRuntime: at android.os.Handler.dispatchMessage(Handler.java:95)
10-16 15:44:57.511 11144 11144 E AndroidRuntime: at android.os.Looper.loop(Looper.java:148)
10-16 15:44:57.511 11144 11144 E AndroidRuntime: at android.app.ActivityThread.main(ActivityThread.java:5417)
10-16 15:44:57.511 11144 11144 E AndroidRuntime: at java.lang.reflect.Method.invoke(Native Method)
10-16 15:44:57.511 11144 11144 E AndroidRuntime: at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
10-16 15:44:57.511 11144 11144 E AndroidRuntime: at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

Depau, 2018-11-03:
I haven't forgotten of this issue, I just wanted to let you know that I've been quite busy and might be able to debug it next week.

I'll tell you though that earlier I think I found what the problem is while waiting for the elevator. Can you check if it works with the [F-Droid build](https://f-droid.org/packages/eu.depau.etchdroid/)? (uninstall the app, download the APK from F-Droid, install it and test it)

If it works with the F-Droid version, uninstall it again and put back the Play Store version. Then with some app backup tool (Titanium, adb, a root file manager, whatever you like) send me the APK.

It may sound like a weird request but I'm suspecting this is due to a new Play Store feature which I enabled. You can now upload to Google an Android Application Bundle instead of an APK. When a user installs an app, Google will mangle the bundle and build a special APK that only contains libraries, artwork and stuff compatible with that device. As I'm not experiencing this issue I'm suspecting Google is removing way too much stuff on low-end devices.

Depau, 2018-11-03:
I updated the Play Store app and, this time, I sent an APK instead of a bundle. The APK will be twice as large but if that's the problem, it should be gone.

I'll put bundles back when I'm sure the problem is fixed.

omkanLoc, 2018-11-06:
Hello Davide,
Thanks for getting back to me, I figured you were busy, so thanks for continuing to work on this.  I dorked up and read your last email first, so I removed EtchDroid and reinstalled thru Play Store before making the APK backup.  sorry for that.  The updated install from the play store still has the same issue/error.  I then uninstalled and tried the F-Droid build.  Had the same crash and error code with that as well.  Would it still be helpful to backup the APK and send it to you at this point?

Depau, 2018-11-07:
@omkanLoc Not really. I'll have to debug the issue. Hopefully I'll find a device on which it occurs.

At least I can now tell it's not Google's fault, I can re-enable the optimized APKs for the next release.

Depau, 2018-11-18:
I checked back on Google Play developer console and noticed the same error occurs on the following devices:

- Galaxy Grand Prime Plus (grandppltedtv)
- CP8676_I02
- LG G3 (g3)
- Galaxy J3 Pop (j3popltetfnvzw)
- Galaxy J7 (j7elte)
- Galaxy A8 (a8elte)
- Aquaris E5 (Aquaris_E5)
- Redmi 3S (land)
- 2 (OnePlus2)
- Xperia Z2 (D6503)
- Galaxy J3 Emerge (j3popltespr)
- Z6
- T5 (T5)
- PB2 plus (PB2-670M)
- Galaxy Note8 (greatlte)
- Desire 10 pro (htc_a56dj_pro_dugl)
- remix_x86
- Galaxy J7 (j7e3g)
- Galaxy S5 (kltetfnvzw)
- Galaxy A7 (a73g)

It also says all these devices are running Android 6.0. You said you were having troubles with Android 6.0.1 so maybe we found the issue. I'll see if I can get Android 6.0.1 on my old Nexus 7 tablet when I get home and try again.

Depau, 2018-11-19:
The bug was a super stupid inconsistency in Android 6.0. I'm releasing v1.3.2, it should be gone :)

closed 2018-11-19 by Depau

omkanLoc, 2018-11-19:
This is great news! thanks for getting this fixed.  I have installed the new version and you are correct the issue is gone.  Just got finished writing an image.

Depau, 2018-11-19:
@omkanLoc It's not completely fixed as I just tested it on KitKat and there's a similar issue, I'll see what I can do later or actually drop support for it. At some point I'll have to drop it anyway in order to support some features I'm planning to add in the future, but it would be nice to keep an old version working on it too.

I certainly cannot support Jelly Bean or earlier as it misses the storage access framework, on which I'm currently relying.

PViswapriya, 2020-12-04:
> Okay, I flashed stock Android 6.0.1 on my Nexus 7 and... I was able to reproduce the bug. I'll now try to troubleshoot it.

hi how you fixed this ?

Depau, 2020-12-06:
> > Okay, I flashed stock Android 6.0.1 on my Nexus 7 and... I was able to reproduce the bug. I'll now try to troubleshoot it.
> 
> hi how you fixed this ?

https://github.com/EtchDroid/EtchDroid/commit/691bddcc0059208e3d25c1cf057b0aca836dd5f4