etchdroid/etchdroid #18: Translations

Depau, 2019-02-04:
I'm opening this issue so all people involved in translating the app can subscribe to it. I'll write here whenever we're good for a [string freeze](https://wiki.gnucash.org/wiki/String_Freeze) and everything can be translated.

Right now I'm planning to do some UI refactoring, which includes a step-by-step wizard to flash USB drives with automatic image type detection + add some more warnings because people who have no idea what they're doing are being [super annoying on Google Play](https://play.google.com/store/apps/details?id=eu.depau.etchdroid&showAllReviews=true).

It's okay if you're not interested in translating the app any more. However if many strings change between one release and the other I may decide to drop the translations in question until somebody picks them back up.

Let me just tag @fyr77 (German) and Exodus1831 (Turkish) so they get a notification for this issue.

# How to translate
## App
1. Open Android Studio, import the project.
2. Once it's done doing all it's loading (it may take a lot), in the code tree expand `app`, `res`, `values`, `string`, open `strings.xml`.
3. It will tell ask you if you want to open the Translation Editor; do it.
4. (only if it's a new translation) In the translation editor, click the button `Add Locale`, select the language
5. Select the language you want to translate and translate all the strings.
6. Repeat steps 2-5 for the file `strings.xml (night)`.
7. Commit your changes (`VCS` → `Commit Changes`) and a pull request.

## Play Store and F-Droid information
1. With a file manager, open the `metadata` directory
2. Create (or copy) a new directory with the language you want to translate the data to
3. Ensure the directory format is exactly the same
  - Changelogs are numbered by the internal Android app version numbered
  - Screenshots don't need to be numbered, but they are sorted in lexicographic order in the stores, so make sure the order makes sense
  - The icon must be the same

yassineim, 2019-02-05:
You might want to use something like https://translate.google.com/toolkit/ to ease the process

Depau, 2019-02-07:
I set up an instance of Weblate: https://etchdroid-l10n.depau.eu/engage/etchdroid/

It's running off some free credit I have on Azure, let me know if it's too slow or it doesn't work. As you can see in the commit history I already tested it and its seems it worked.

Hopefully it will be usable, both in terms of performance and user friendliness.

yassineim, 2019-02-08:
Hello, I just signed up, not slow at all.
You might want to add French

Depau, 2019-02-08:
@yassineim Done ;)

Depau, 2019-02-08:
Add yourself [here](https://github.com/EtchDroid/EtchDroid/blob/master/app/src/main/res/xml/about.xml#L122) and send a pull request so you show up in the translators page :)

yassineim, 2019-02-08:
Cool but I'm ashamed to do that before having translated anything haha

c-sanchez, 2019-05-04:
Hi @Depau I want translate the app in Spanish, please add Spanish in the list to make the translation :)
Curiosly the EtchDroid Translations page is in spanish completely hah.

Depau, 2019-05-06:
Hi! Sorry for having you wait, here you go: https://etchdroid-l10n.depau.eu/projects/etchdroid/app/es/
Let me know if you think I should also start translations for Spanish variants.

c-sanchez, 2019-05-07:
Thank you very much @Depau :D
About spanish variant I think is not necessary, at least not until someone else wants to make their own variant, I think my translation is normally "spanish international" :P
So just -spanish- is ok for me :)

guland2000, 2019-06-27:
Russian is done.

Depau, 2019-10-18:
The translations website has been moved to a Weblate-hosted instance of @weblate at https://hosted.weblate.org/projects/etchdroid, which was kindly provided for free by the company behind it.

I set up redirects on the old server, so all your previous links should continue to work.

Sadly you'll have to create new accounts.

Thank you very much for your contributions!

comradekingu, 2020-01-12:
@Depau Why not make the relevant section of the about page into a string?

Depau, 2020-01-12:
@comradekingu what are you referring to, exactly? I'm not sure what you mean

comradekingu, 2020-01-13:
@Depau Firstpost details adding oneself to AUTHORS here. Instead the string for translator of current locale (the only interesting feedback-loop), could be a translatable string.

Depau, 2020-01-13:
I think it's better to have all the translator credited, not just the one for the current translation :)

Also, though it can be certainly done, the Attribouter library I'm using makes it easier to do it the way it's currently done.

Depau, 2020-07-18:
I finally have some free time to dedicate to this project. First of all I wanna thank all the people that contributed translations.

I have quite a few translations that have been contributed but have not been released. I will do a release around next week.

I'm letting you know in case you guys want to give a look at your language and finish it up.

Today I made using Weblate mandatory: since hand-made translations XMLs tend to be quite troublesome to carry around when performing merges, I added a validation script that soft-rejects such pull requests.

Also, I added some plugins to Weblate which should provide Google Translate suggestions and perform an auto cleanup when you guys are done. Let me know if you notice any issues :)

comradekingu, 2020-09-07:
1. @Depau I don't think you can use Google Translate, as per their terms. Doing so is in any event a bad idea, as the quality isn't good. Additionally it might be perceived as the translations stem from the the respective translators in the AUTHORS file.

2. The license agreement on EtchDroid Weblate is not needed, as it is a duplicated of 6.4 in https://hosted.weblate.org/legal/terms/

3. The license is GPLv3-only in https://hosted.weblate.org/projects/etchdroid/app/nb_NO/#information
whereas I think the project is GPLv3+

Depau, 2020-09-12:
>     1. @Depau I don't think you can use Google Translate, as per their terms. Doing so is in any event a bad idea, as the quality isn't good. Additionally it might be perceived as the translations stem from the the respective translators in the AUTHORS file.

It is indeed paid anyway. The idea was not to translate the app automatically but provide machine-translated translations to human translators.

>     2. The license agreement on EtchDroid Weblate is not needed, as it is a duplicated of 6.4 in https://hosted.weblate.org/legal/terms/

I added it as a reminder to translators of what license their translations will be licensed under. It is not required but it's a useful reminder imo.

>     3. The license is GPLv3-only in https://hosted.weblate.org/projects/etchdroid/app/nb_NO/#information
>        whereas I think the project is GPLv3+

The project is GPLv3, it is indeed inconsistent on F-Droid. I'm not sure what to do about it since I have accepted a bunch of contributions already and project-wide its reported as just "GPLv3". For me it's fine either way as long as it's a copyleft license.

Thanks for spotting that!

comradekingu, 2020-09-12:
@Depau 
Automated suggestions (as they are now called) are there anyway.
The license is already stated, no need to reiterate with a required CLA. If anything the "this requires you to accept …" will make translators drop the project on sight.
Since the license was in error, and multiple projects have done it, I think it is fine to change the license.

Depau, 2020-09-12:
> @Depau
> Automated suggestions (as they are now called) are there anyway.
> The license is already stated, no need to reiterate with a required CLA. If anything the "this requires you to accept …" will make translators drop the project on sight.

I'm not begging for anyone's contributions, also I don't want to deceive anyone on the license. If someone doesn't want to acknowledge that their translations will have to be licensed under the GNU General Public License version 3.0 and everything that the license implies, then I'm totally fine with them not translating the app.

It is not a binding contract or a "customer agreement", it is a reminder that the app is open source and GPL3 licensed, and so will be their translations if they want to contribute them.

I really don't see what's the fuss about this, yeah I could turn it into an announcement but what's the difference? I'm not requesting or expecting anything from anyone. I'm just trying to be transparent, maybe someone doesn't like Richie's license and won't translate it because of that, and if that's the case I'd rather remind them before translating rather than deceive them into translating the app before they realize it.

Depau, 2020-09-12:
## Actual translation/project update

The new libaums release which includes my fixes for emulated and other weird USB devices is out, so it's time to do a release, which I'll hopefully have time to do within the next 2 weeks.

This should substantially kill 99.999% of software-caused USB bugs, therefore better stability :tada: 

I'm not gonna do any UI additional GUI/string changes, so all translations work can go on undisturbed without fears. Remember to credit yourself in the about page, https://github.com/EtchDroid/EtchDroid/blob/develop/app/src/main/res/xml/about.xml

When I'll push the release I'll also take care of the license inaccuracy on F-Droid.

On the "About" page, it was brought to my attention that some languages are not detected (for example Russian). That's a bug in Attribouter that needs to be fixed upstream, the problem is that Android additionally requires the country code for some language to be properly recognized (for example `ru_RU` as opposed to just `ru`, which instead works for most countries). However, Attribouter does not support this syntax.

This is also a shout-out to anyone who wants to give a look into it and send a pull request to Attribouter, I'll pull in any updates as they come.

yarons, 2023-12-28:
Hi, the Translation link on F-Droid still points to the Azure Weblate you've provisioned.

depau, 2024-01-08:
> Hi, the Translation link on F-Droid still points to the Azure Weblate you've provisioned.

Thanks for reporting! I opened a merge request for the F-Droid data, and I also set up a redirect from the old domain, it should be online soon.

Shipsea-a, 2025-03-08:
I recently polished the Chinese translations for the project. I'd like to be listed as a contributor. Please let me know if any further information is needed.