etchdroid/etchdroid #20: Error attaching to device

radiatehopecfh, 2019-03-08:
I have a screenshot of the error info. it's telling me to reboot or restart device. I'm going to try that now, hopefully that works. I really needed this to work quickly not cause me more obstacles in my way!
[image]

Depau, 2019-03-08:
@radiatehopecfh You probably just moved the adapter a little bit and it disconnected from the phone for a small fraction of time.

Or you're running Android 9, on which this is a known bug.

In one of the next release I would like to add the ability to resume flashes, it should be helpful in situations like these.

For the time being just try to leave the phone alone on a table so you're sure you don't accidentally disconnect anything.

closed 2019-03-08 by Depau

Depau, 2019-03-08:
BTW feel free to reopen it if your issue is different.

Depau, 2020-01-13:
Hi @radiatehopecfh, if you're still having the issue, can you test the latest version and see if it's working fine now? It should be doing a lot better.