etchdroid/etchdroid #22: Just wanted to post this error I keep getting trying to write lubuntu to usb. Butt has worked great on pretty much everything else

kkolstad72, 2019-03-20:
Write failed at block 3309568, 1.7 GB written. Error: java.io.IOException: Unsuccessful Csw status: 1

Depau, 2019-04-03:
Hello, sorry for not having replied earlier, I've been quite busy. The issue you're reporting is quite interesting, I've never seen that exception message in particular, can you tell me more about your setup (Android version, device model)?

kkolstad72, 2019-04-03:
I have an lg g7 thin Q
Which could be the problem. It's a piece of crap.
Thanks for responding

Depau, 2019-04-28:
Again, sorry for the late answer, I've been busy.

Can you please tell me the Android version? To rule out an hardware issue you could try copying a file to the drive using the built-in file manager, see if it fails.

zoomer296, 2019-07-15:
Well, they've disappeared. The G7 ThinQ ran Oreo at the time of their bug report, but has recently been updated to Pie.

Depau, 2019-08-10:
Alright, congratulations, you'll now hit the Android Pie bug #10 :tada: 

As an update I'm currently working on rewriting the back-end part. I'll first try to remove all the shitty code and duct-tape it to the existing UI, then I'll try a few workarounds I have in mind for that stubborn bug.

Depau, 2019-08-10:
But wait, @zoomer296, are you saying that you used to have the same issue @kkolstad72 had but now it works on Android Pie?

zoomer296, 2019-08-10:
Sorry, I'll state that more clearly. 
The user with the issue stopped responding. 
Their phone only had 8.0 available at the time of the bug report, so that rules out the Pie bug as the cause of their issue, but future issues with this particular phone may be Pie-related.

Their issue is probably because of LG's heavily-modified kernel.

Depau, 2019-08-10:
Okay, thank you.

I don't think it's the kernel's fault. The issue only and always occurs on Android 9, which rules out any kernel issues as different devices have completely different kernel versions.

Depau, 2020-01-13:
Hey @zoomer296 @kkolstad72,
can you try with the latest release? There has been a lot going on in libaums, I expect it to be much more reliable.

zoomer296, 2020-01-13:
Will do. I should be able to get to it sometime today.

Depau, 2020-05-21:
Hi, I'm getting back to this to do some issue clean up. I merged an updated libaums quite a while ago that uses `libusb` instead of the official Android API to access the USB devices.

That doesn't fix this kind of issue in every case but according to Play Store reviews and my experience it has improved things by a lot.

If you try again now, are you getting any issues?