etchdroid/etchdroid #23: app crashes

c-sanchez, 2019-05-04:
EtchDroid crashes selecting any image, and opening the about info with dark theme selected too.
Theses crashes occurs with my tablet, android 4.4.4 kitkat, intel x86

Depau, 2019-05-06:
@c-sanchez Hi! Can you send crash logs please?

Thank you

c-sanchez, 2019-05-07:
Hi @Depau, I'm not sure about where is the crash logs, since this is a forced close, I can't view any, crash log per se.
Anyway I taked an logcat with **Logcat Extreme** app, then I taked the logs just from EtchDroid.
I will give attached two log files, one is the when I change the app theme to dark and then opening the about menu, curiously this don't occurs with the normal theme.

The other log is when select the image.

I'm not sure how useful can be this, if you need an more detailed crash log or something with other program/method let me know :)

Also here an video from my tablet using EtchDroid and getting the "app closes forced"
[https://www.mediafire.com/file/wpwxi0x1vsqgi3w/VID_20190506_230732.mp4/file](url)

Best regards.

[LogcatX.zip](https://github.com/EtchDroid/EtchDroid/files/3153543/LogcatX.zip)

Depau, 2019-06-13:
Just a quick update. I've seen your reply, I just haven't found the time to work on it.

I'll hopefully get on it in around a month.

Depau, 2020-01-13:
Hey @c-sanchez,
sadly I decided to drop support for Android < 5.0, there are too many issues and inconsistencies with it. It will not be supported any more unless somebody helps me fix it. Let me know if you're still having issues on newer versions of Android.

closed 2020-01-13 by Depau

c-sanchez, 2020-01-13:
> Hey @c-sanchez,
> sadly I decided to drop support for Android < 5.0, there are too many issues and inconsistencies with it. It will not be supported any more unless somebody helps me fix it. Let me know if you're still having issues on newer versions of Android.

A shame, but I understand, its ok, I will comment if I found something in newer Android versions then :)
Regards.