etchdroid/etchdroid #25: Etch Crashes - Android 4.4.2: Etch cannot see the .iso

Rbagman, 2019-06-11:
I figure lack of permissions is why I can't navigate to the iso file. 

When, I use ghost commander/oi file manager/amaze to open the file Etch and vlc are the two options. Etch crashes while starting.

java.lang.RuntimeException: Unable to start activity ComponentInfo{eu.depau.etchdroid/eu.depau.etchdroid.activities.UsbDrivePickerActivity}: android.view.InflateException: Binary XML file line #25: Error inflating class TextView
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2413)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2471)
	at android.app.ActivityThread.access$900(ActivityThread.java:175)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1308)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:146)
	at android.app.ActivityThread.main(ActivityThread.java:5602)
	at java.lang.reflect.Method.invokeNative(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:515)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1283)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1099)
	at de.robv.android.xposed.XposedBridge.main(XposedBridge.java:132)
	at dalvik.system.NativeStart.main(Native Method)
Caused by: android.view.InflateException: Binary XML file line #25: Error inflating class TextView
	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:719)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:761)
	at android.view.LayoutInflater.rInflate(LayoutInflater.java:769)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:498)
	at de.robv.android.xposed.XposedBridge.invokeOriginalMethodNative(Native Method)
	at de.robv.android.xposed.XposedBridge.handleHookedMethod(XposedBridge.java:631)
	at android.view.LayoutInflater.inflate(Native Method)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:398)
	at android.view.LayoutInflater.inflate(LayoutInflater.java:354)
	at androidx.appcompat.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:469)
	at androidx.appcompat.app.AppCompatActivity.setContentView(AppCompatActivity.java:140)
	at eu.depau.etchdroid.activities.UsbDrivePickerActivity.onCreate(UsbDrivePickerActivity.kt:103)
	at android.app.Activity.performCreate(Activity.java:5451)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1093)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2377)
	... 12 more
Caused by: android.content.res.Resources$NotFoundException: File res/drawable-night-v8/ic_usb_200dp.xml from drawable resource ID #0x7f080094
	at android.content.res.Resources.loadDrawable(Resources.java:3422)
	at android.content.res.TypedArray.getDrawable(TypedArray.java:602)
	at android.content.res.XResources$XTypedArray.getDrawable(XResources.java:965)
	at android.widget.TextView.<init>(TextView.java:1233)
	at androidx.appcompat.widget.AppCompatTextView.<init>(AppCompatTextView.java:87)
	at androidx.appcompat.widget.AppCompatTextView.<init>(AppCompatTextView.java:83)
	at androidx.appcompat.app.AppCompatViewInflater.createTextView(AppCompatViewInflater.java:177)
	at androidx.appcompat.app.AppCompatViewInflater.createView(AppCompatViewInflater.java:102)
	at androidx.appcompat.app.AppCompatDelegateImpl.createView(AppCompatDelegateImpl.java:1266)
	at androidx.appcompat.app.AppCompatDelegateImpl.onCreateView(AppCompatDelegateImpl.java:1316)
	at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:690)
	... 26 more
Caused by: org.xmlpull.v1.XmlPullParserException: Binary XML file line #1: invalid drawable tag vector
	at android.graphics.drawable.Drawable.createFromXmlInner(Drawable.java:986)
	at android.graphics.drawable.Drawable.createFromXml(Drawable.java:930)
	at android.content.res.Resources.loadDrawable(Resources.java:3418)
	... 36 more

Depau, 2019-06-13:
Hi,
The issue is caused by an Android <4.4 limitation. Android versions before Lollipop do not support vector graphics out of the box.

I tried to keep support for Android 4.4 but there are too many differences between that version and newer ones, so I'll drop support for it with next release unless somebody is willing to send patches.

I'll keep this issue open so I remember to raise the minimum Android version required to Lollipop.

c-sanchez, 2019-06-17:
Maybe avoid the vector graphics usage?
At least an "special build" for android 4.4 users, like me..

Depau, 2019-06-17:
@c-sanchez Feel free to send a pull request. It should be quite doable with compatibility library (which I'm pretty sure are already used, I might not be doing it right in that particular screen).

I'm simply not interested in it because 1) I haven't had any device running KitKat for years 2) it's like 1% of the user base.

But if anybody sends good patches I'll sure merge them.

Depau, 2019-06-17:
Relevant code is in this layout: https://github.com/EtchDroid/EtchDroid/blob/master/app/src/main/res/layout/activity_usb_drive_picker.xml

As you can see there's a TextView with an embedded graphic `drawableTop` (which is vector and that doesn't work on Android <5).

You should be able to do something like this: https://github.com/Depau/POuLBITS-Android/blob/master/app/src/main/res/layout/activity_main.xml#L31

That is convert the standalone TextView to a ConstraintLayout (or another layout, whatever you prefer) containing an ImageView and a TextView _without_ the image. In the ImageView you should be able to use the attribute `app:srcCompat` instead of `android:src`, which will be picked up by AndroidX vector compatibility libraries.

It will generate a non-vector version of the drawable that will be used instead of the vector version on older versions of Android.

Note that I enabled the Google Play Store option that automatically reduces the size of the APK when possible, so the non-vector image will only be downloaded by people using older versions of Android.

Anyway, beware that I might still think of dropping support for Android KitKat for some more reasons, namely:

- Android Storage Access Framework was introduced in that particular version (which is why I don't support earlier versions). It behaves a little bit differently than >4.4 so it might be a pain
- Notifications also work a bit differently. In fact IIRC I wasn't able to get the progress bar on Android 4.4 (relevant code [here](https://github.com/EtchDroid/EtchDroid/blob/master/app/src/main/java/eu/depau/etchdroid/services/UsbWriteService.kt#L44 and following lines). You may want to look into that too.

As I mentioned in my [development update](https://blog.depau.eu/2019/05/18/etchdroid-development-update/) in my blog I will be able to finish later this summer the changes I started in February, currenty pending in the [`development` branch](https://github.com/EtchDroid/EtchDroid/tree/develop). If I find some way to test it on KitKat with my current hardware I definitely will keep an eye on it.

I would also like to implement instrumented integration tests which would be VERY helpful for problems like this. And that's going to suck because I have no idea how to emulate USB drives in a Travis CI-like environment. I'll keep you posted though.

Depau, 2019-06-17:
Oh and, by the way, I'm quite sure that if you plug in the USB drive before starting the app, Android won't even try to render the drawable so it should not crash. That can be a nasty workaround for the time being lol

Depau, 2020-01-13:
I eventually dropped support for Android 4.4 because of all this little issues and inconsistencies. 5.0 and up will only be supported, that is unless somebody steps in and helps me fix it. I'll close this issue for now.

closed 2020-01-13 by Depau