etchdroid/etchdroid #26: Error in writing  to usb in android pie version

SureshKumar311, 2019-06-12:
When I am flashing with raspberry full image OS it was struck and become error of

Write failed at block 241664, 123.7 MB written. Error: java.io.IOException: Could not write to device, result == -1 errno 88 Socket operation on non-socket 

So please help me

SureshKumar311, 2019-06-12:
I had reboot my phone also and same problem is they

SureshKumar311, 2019-06-12:
An another try error 
Write failed at block 2727936, 1.4 GB written. Error: java.io.IOException: Could not write to device, result == -1 errno 88 Socket operation on non-socket

Depau, 2019-06-13:
Duplicate of #10.

This is a known limitation on Android 9. I still haven't figured out what causes the issue.

closed 2019-06-13 by Depau

Depau, 2020-01-13:
Hi @SureshKumar311, can you test the latest version and see if it's working fine now? It should be doing a lot better.