etchdroid/etchdroid #27: EtchDroid, "Could not access USB device."

whunder, 2019-06-24:
*OnePlus 5T
*Android Pie 9.0 (Build number: ONEPLUS A5010_43_190527)
*USB devices: Ugreen USB-C/USB3.0 to SD-Card adapter (https://www.amazon.com/UGREEN-Reader-Adapter-Portable-Windows/dp/B07D1J88CF), Samsung EVO+ 32Gb SD-Card (Formatted on Android prior to doing anything).
*Turned on OTG Storage in Android settings prior to opening app.
*Gave EtchDroid Storage permissions prior to opening the app.
*ISO is openmediavault for Raspberry Pi.
*Writing method being used is: Write image directly to disk (using Android API).

*After seeing error for the first time, I rebooted and tried everything again before doing anything else.

*Full disclosure, I am rooted (with Magisk) on stock ROM.

Depau, 2019-06-24:
@whunder Hi, can you send a screenshot of the error message? Have you tried to close the app and try again?

whunder, 2019-06-24:
[image]
 Yes, I restarted the app and the phone one at a time. I also reset the error messages just in case it was a 'false positive'.

Depau, 2019-06-26:
Interesting. This usually happens when you remove the USB drive and put it back in after you've authorized the app to access it. The app thinks it already has access but it actually does not.

Usually this is solved by closing the app and swiping it away from the recent tasks screen.

If it doesn't work this way it might be a bug and I need a log. Do you know how to record a logcat?

This should grep only the app logs if you run it while the app is running and store it in your home directory (works on GNU/Linux and macOS)
```
adb logcat -d | grep $(adb shell ps | grep 'eu.depau.etchdroid$' | awk '{ print $2 }') > ~/etchdroid-log.txt
```

Remove any sensitive information there might be, upload it to a pastebin such as hastebin.com and post the link here.

whunder, 2019-06-26:
Thank you. I will execute that tonight. Where would you like me to upload the resulting file?

Depau, 2019-06-26:
@whunder Whatever is good for you, hastebin or gist are fine

whunder, 2019-06-26:
Here you go. I had to switch back and forth with the USB cable and the USB drive but I think everything was captured. If you need me to do that from a terminal emulator on the phone, let me know. I'll do what I can.

[Pastebin paste](https://pastebin.com/g2YJsaei)

Depau, 2019-08-10:
Alright, sorry for getting back to you after a month. Thank you very much for the logs, they're kinda... interesting? The logs look like nothing bad happened.

Anyway I'm currently reworking the back-end to clean it up a bit, I'll get back to you after that and if you're available I'll have you do some tests.

whunder, 2019-08-10:
Lol. I do understand. As a hardware engineer, I do understand being pulled in multiple directions. I will be happy to test anything you come up with. I would like to divulge that OnePlus (I have the One Plus 5t) did take liberties with the USB 3.1 plug spec and may have broken something when designing it.

Depau, 2019-08-14:
As long as they (re)implemented the USB access framework it should work just the same.

I do have a OnePlus 3T (actually I *did* because I broke the screen last week) and I had been able to use EtchDroid just fine until I switched to a Pie custom ROM. I also tested it on my computer on Android-x86 and on my Nexus 7 2013, still perfect.

I'll have to see the code and git history in the Android framework, it will take a lot of time to find the regression.

But anyway I'd say I'm 70% through the back-end rewriting part so we'll know soon :)

Depau, 2020-01-13:
Hi @whunder,
can you try with the latest release? The guy behind libaums did a lot of work, many issues should be gone now.

whunder, 2020-01-14:
Hi @Depau,

Latest release works like a charm! Thank you for keeping on this! You have earned a couple of coffees (coffee!?) from me!

Now if you implement writing zipped iso files...! Lol.

Daniel

closed 2020-01-14 by whunder

Depau, 2020-01-19:
Hi,
I'm glad it works, thanks for the many coffees! :)