etchdroid/etchdroid #29: Brazilian Portuguese translation is broken, empty

rffontenelle, 2019-08-10:
Somehow, [app/src/main/res/values-pt-rBR/strings.xml](https://github.com/EtchDroid/EtchDroid/blob/master/app/src/main/res/values-pt-rBR/strings.xml) was created (in [4e8c0c42](https://github.com/EtchDroid/EtchDroid/commit/4e8c0c423eef81065e297a6614c65900a41941b8)) with the content:
```
<?xml version="1.0" encoding="utf-8"?>
<resources></resources>
```
This also makes impossible to contribute via weblate

Depau, 2019-08-10:
Hi! I'll fix it, give me a minute

Depau, 2019-08-10:
It actually works, see this screenshot:

[image]

Just go to the [Brazilian Portuguese translation page](https://etchdroid-l10n.depau.eu/projects/etchdroid/app/pt_BR/) and click the `Translate` button on the right:

[image]

Let me know if you have any issues.

closed 2019-08-10 by Depau