etchdroid/etchdroid #30: Can not boot on the USB key.

JordanChr, 2019-08-14:
Hello,

I wanted to know on what format the bootable USB key is made?  NTFS, FAT32?

Depau, 2019-08-14:
The image you provide is copied as it is to the USB drive, overwriting any file system in it just like [Etcher](https://etcher.io) or dd would.

Many of Etcher's [FAQ](https://www.balena.io/etcher/#faq) also apply to EtchDroid.

Are you trying to flash a Windows image? It's not supported.

JordanChr, 2019-08-14:
Yes, I try to flash windows 10. I would like to repair my PC but can not
boot to the USB key.  I wonder if this could not come from the key format
surely in FAT32.  My PC is 9 years old, I think it can only support the
NTFS.

Depau, 2019-08-14:
EtchDroid does NOT support Windows installers. Read #5

Depau, 2019-08-14:
You can use WoeUSB on GNU/Linux: https://github.com/slacka/WoeUSB/

closed 2019-08-14 by Depau

JordanChr, 2019-08-14:
The problem with WoeUSB is that it works from a PC and that I only have an
android phone to repair my PC.

Depau, 2019-08-14:
Then I'm sorry but EtchDroid can't help you.

You can still flash, for example, an Ubuntu ISO, boot from it and then use WoeUSB on Ubuntu to create a Windows installer on another USB drive.

Depau, 2019-08-14:
If your phone is rooted you can also try DriveDroid but your mileage may vary.

JordanChr, 2019-08-14:
My phone is not rooted, but I will try ubuntu.  Thank you for your help.😀

JordanChr, 2019-10-29:
Thank you for your help, my PC is repaired. 😉

(And I kept the Ubuntu bootable USB stick I created with EtchDroid)