etchdroid/etchdroid #35: [Bug] systematically stops at ~50%

bphd, 2019-09-28:
whatever I try to do it stops
Zenfone 6 on Pie

You really should stop the notification thing and just let the app open during the process, it will resolve a lot of problems because Android will not shut down the process

Depau, 2019-09-29:
Duplicate of #10, please add further comments there.

The problem is not related to the notification. The problem is that [libaums](https://github.com/magnusja/libaums/) does not yet support recovering from a particular error condition which, for whatever reason, happens always and consistently on Android Pie.

We're pretty sure this is what causes the issue, as we've determined other closed-source apps that try to accomplish a similar goal behave in the same way except for this detail. Please check https://github.com/magnusja/libaums/issues/209 for further info.

Depau, 2019-09-29:
For what it matters, next version of EtchDroid will also show a progress activity unless you close it, and in general an improved UI. Still, this is not what is causing the issue.

closed 2019-09-29 by Depau

Depau, 2020-01-13:
Hi @bphd, can you test the latest version and see if it's working fine now? It should be doing a lot better.