etchdroid/etchdroid #36: [Bug]

ayushkssk, 2019-11-24:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: [e.g. OnePlus 5]
 - OS: [e.g. Android 8.0]
 - Version [e.g. 22]
 - Installed from: [Play Store, F-Droid, GitHub, self-built...]

**Additional context**
Add any other context about the problem here.
Failed each time i tried

Depau, 2019-12-04:
Hi, can you please explain a little better what's the problem?

Thank you.

Depau, 2020-01-13:
Hi, I still haven't heard from you. Just as a heads up, try the latest release and see if it works. If it doesn't, let me know but please explain yourself better, otherwise I don't know how to help you.

closed 2020-01-13 by Depau