etchdroid/etchdroid #37: Stuck when enabling night mode

glitchphan, 2020-01-19:
Latest version from F-Droid on Umidigi X (China phone) with Android 9. When I hit enable night mode, the screen goes black, back button does nothing, and trying to switch back to it with the switcher doesn't work (tapping on it does nothing). Force closing it and opening it again also just gets back to black screen and hung state. Have to uninstall and reinstall to start again.

Depau, 2020-01-20:
Hi,
I'm painfully aware of this. The latest version is supposed to have it disabled on Android 9 or newer, I don't know if you updated it.

Basically it conflicts with the system night mode setting, so if you change it it goes on in a loop trying to reload the activity forever.

As I said, though, it's supposed not to be controllable by the user in the last version. Can you please double check?

Clearing the app storage from system settings should work around it for now.

glitchphan, 2020-01-21:
Thanks, it was the 1.3.4 version in F-Droid that I was using before. Today I updated to 1.4 and that option is no longer there.

depau, 2025-05-02:
Night mode reimplemented with Jetpack Compose

closed 2025-05-02 by depau