etchdroid/etchdroid #38: [Bug] USB drive remounted automatically by Android

linuxandria, 2020-01-20:
**Describe the bug**
While EtchDroid is initializing the writing, Android remounted the USB automatically after EtchDroid formats it. The wrote then fails, because after remount Android doesn't recognize the filesystem in the drive and doesn't make it available to apps.

**To Reproduce**
Steps to reproduce the behavior:
1. Try to write Ubuntu ISO
2. Failure

**Expected behavior**
Not that

**Smartphone (please complete the following information):**
 - Device: Pixel 2XL
 - OS: Android 10, January 2020 patch
 - Version: 1.4
 - Installed from:  Play Store

**Additional context**
Add any other context about the problem here.

linuxandria, 2020-01-20:
I am rooted so if there's a workaround that way...

Depau, 2020-01-20:
@linuxandria Hi, good that you're rooted, can you attach a logcat and a kernel log (dmesg)?

Thank you

Depau, 2020-03-14:
Hi, since I haven't heard from you I'll close this issue. If you're still experiencing the problem, let me know and I'll reopen it.

closed 2020-03-14 by Depau