etchdroid/etchdroid #39: Add sdcard support without using the adapter

DinoWattz, 2020-01-26:
**Is your feature request related to a problem? Please describe.**
I use a microsd adapter for pc and my phone does not work with otg

**Describe the solution you'd like**
I wanted the app to work with a sd card without an adapter

Depau, 2020-01-26:
Hi,
this is not possible. I'm sorry.

closed 2020-01-26 by Depau

DinoWattz, 2020-01-26:
> Hi,
> this is not technically possible. I'm sorry.

Even with root?

Depau, 2020-01-26:
Yes, it's totally possible with root, however:

- it needs a whole lot of workarounds and detection routines for different devices
- it needs quite a lot of double-triple-quadruple-checks to avoid bricking phones (and no, I'm not gonna pay back anybody cause they bricked their phone with my app)
- it needs a lot of testing to ensure OEM ROMs don't simply try to use the SD card while the app is working
- root-based functionality is out of the scope of this project

So no, this is not going to be implemented, not even in the future. Either get a better USB-OTG SD card adapter or write an app that does that with root.

BTW, if you have root you can easily use `dd` on Termux, you don't really need an app for that. It's usually `/dev/mmcblk-something`, be careful not to write it on your phone's memory.