etchdroid/etchdroid #40: Write failed at block 2828288

mavaddat, 2020-01-26:
Got an error:
```
Write failed at block 2828288, 1.4 GB written. Error: java.io.IOException: libusb returned -1 in control transfer
```
Was trying to write `Windows10_InsiderPreview_EnterpriseVL_x64_en-us_18363.iso`
That is
Windows 10 Enterprise VL Insider Preview - Build 18363 English

64-bit Download

Depau, 2020-01-26:
> Was trying to write Windows10_InsiderPreview_EnterpriseVL_x64_en-us_18363.iso
> That is
> Windows 10 Enterprise VL Insider Preview - Build 18363 English

See #5 

[image]

[image]

I don't even know how to say it, why can't people read 😩😩😩

---

On to the issue itself, when you clicked "New issue" you were presented with a list of information you should have provided. Most importantly:

> **Smartphone (please complete the following information):**
>  - Device: [e.g. OnePlus 5]
>  - OS: [e.g. Android 8.0]
>  - Version [e.g. 22]
>  - Installed from: [Play Store, F-Droid, GitHub, self-built...]

Please provide that and full logs, not just the last line.

mavaddat, 2020-01-26:
My apologies. Actually, I did read that line and assumed it implied application capability (of writing all kinds of sophisticated ISO files) rather than limitation. I thought it meant it can do _all the good stuff._

Anyway, after I posted this issue, I was able to **_successfully_** write the Windows ISO mentioned above by forcibly "pinning" the EtchDroid app so that Android wouldn't allow the USB drive to be swapped by the system (retain priority with EtchDroid).

My system information is as follows:
```
Android 10 / API 29
Baseband Version: g8150-00041-191016-B-5945070
Build Number: QQ1B.200105.004
Linux Kernel Version: 4.14.111-ge58a32340f44-ab6027802
Http User Agent: Dalvik/2.1.0 (Linux; U; Android 10; Pixel 4 Build/QQ1B.200105.004)
```
I'm using EtchDroid 1.4, updated Jan. 12, 2020
Installed EtchDroid from Playstore.

Full bugreport-flame-QQ1B.200105.004:
[bugreport-flame-QQ1B.200105.004-2020-01-26-09-01-44.zip](https://github.com/EtchDroid/EtchDroid/files/4113790/bugreport-flame-QQ1B.200105.004-2020-01-26-09-01-44.zip)

mavaddat, 2020-01-26:
The USB jump drive is Philips USB Flash Drive 13fe:6300. FM32FD00B

Depau, 2020-01-26:
Thanks for the information.

Yes it looks like the issue was simply due to a temporary disconnection of the USB drive (maybe you touched it and it disconnected for a fraction of a second), or Android was simply being a bitch and killed the app.

I suggest you add the app to the "not optimized" list in the battery settings if you continue experiencing. The app will not run in any case until you start it, so it won't hurt battery life.

Try that and let it sit on a still surface while it's working. If it still occurs, feel free to open another issue.

closed 2020-01-26 by Depau

mavaddat, 2020-01-27:
Thank you! Right after my last comment, the software successfully created a bootable UEFI USB drive for Linux Ubuntu 19.10 `eoan` to the USB. I am very impressed with this software and I will commit myself to contribute meaningfully to this project.

Depau, 2020-01-27:
@mavaddat Thank you!