etchdroid/etchdroid #41: Screenshot links might be dead

TomasHubelbauer, 2020-01-31:
Hi, the screenshot thumbnails to the readme lead to links which appear dead to me at least on the network I'm trying to access them from. May I suggest placing the full size images into the repo and using `<img` attributes with `width` set instead in the readme? That way they can just link to themselves and GitHub will show the full version on click.

Depau, 2020-02-02:
Hi,
the images work for me:
[image]

All screenshots are already in the repo: https://github.com/EtchDroid/EtchDroid/tree/master/metadata/en-US/images

If you mind sending a pull request with the screenshots from the repo but with a small size that'd be great, I fighted a lot with the markdown which is why I eventually used postimage.org.

TomasHubelbauer, 2020-02-02:
I see. It was probably just a quirk of the network I was on. Still I think removing PostImg is worth it if the images are already in the repo. I will send the PR for that. I'll use normal HTML image elements embedded in Markdown instead of the Markdown syntax because it doesn't support sizing the images.

PR #43 (2020-02-03): Use images directly from the repo
Closes #41.

I don't know why, but when I used this syntax:

```md
<img>
<img>
<img>
```

The images appeared one on top of another even though they should have showed up inline. Definitely weird.

I "fixed" it by appending an HTML comment after each one which resulted in the expected layout.
I have no idea if this is a GFM issue or just a MarkDown thing in general.

In any case, it's no big deal IMO and now you can drop a dependency on a 3rd party image hoster.
closed 2020-02-03 by Depau