etchdroid/etchdroid #49: [Bug]

imyungdae, 2020-03-14:
**Describe the bug**
(Possibly the same as issue 38)
I select Linux iso, select usb, write starts, app closes or CTDs with "check notification" message. It goes a couple percents then nothing, the notification disappears and soon i get "unsupported usb drive" message.

**To Reproduce**
Steps to reproduce the behavior:
As above:
1. Click on "select iso"
2. Click on "select usb"
3. See error, process seemingly stops.

**Expected behavior**
Iso written to my usb.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: Xiaomi Redmi 7A
 - OS: Android 9
 - Version: PKQ1
 - Installed from: Play Store

**Additional context**
In my case, phone is NOT rooted.

imyungdae, 2020-03-14:
[image]

Depau, 2020-03-14:
Hi,
when the write stops you should get an error notification. If you open it it should tell you what happened, could you please attach that?

Also, if you're running MIUI it's quite likely it's messing with EtchDroid due to its stupid power restrictions. Could you try following the instructions in the following link to disable the power restrictions? (I promise it won't drain your battery :) )

MIUI: https://dontkillmyapp.com/xiaomi
Android One or custom ROM: https://dontkillmyapp.com/google

In addition, try leaving the device on a table while it's working, don't touch it. Some USB adapters are bad and make poor contact, movement may cause temporary disconnections from which the app can't recover yet.

imyungdae, 2020-03-14:
I tried looking into those settings but my MIUI version seems to be newer and some settings are not there and couldnt find. I managed to disable power restriction for the app, as well as granting some more permissions just in case. I tried rebooting and reinstalling, problem still persists.
Also I noticed that the background is eifferemt than on your screenshot on the Play Store.
[image]

imyungdae, 2020-03-15:
On a side note, is there a possibility that app somehow switched into night mode and thats why it looks like this? It would be weird since im not using dark theme on my phone and I did not enable it in EtchDroid app. In fact I didnt even find such setting.

Depau, 2020-03-16:
That looks weird, try clearing the app's storage.

Anyway that's not the screenshot I was looking for. I was looking for a screenshot of the error **after** it fails, which you can get by tapping on the notification.

imyungdae, 2020-03-16:
In the meanwhile the problem got solved, I managed to write. I just put my phone on the table as you suggested and REALLY carefully touched the screen to select the iso file and the device.

closed 2020-03-16 by imyungdae

Depau, 2020-03-16:
Alright, I'm glad it worked. Consider getting a new adapter, they tend to get bad quite quickly, especially the micro USB ones :)