etchdroid/etchdroid #51: [Bug] Write failed at block 2449408, 1.3 GB written. Error: java.io.IOException: libusb returned -7 in control transfer

emayljames, 2020-05-17:
"Write failed at block 2449408, 1.3 GB written. Error: java.io.IOException: libusb returned -7 in control transfer"

**Describe the bug**
Select Ubuntu 20.04 iso -> plug in usbc drive -> accept messages -> gets to at least beyond 50% then fails.

**Smartphone (please complete the following information):**
 - Device: [Xiaomi Redmi Pro]
 - OS: [Android 6.x]
 - Installed from: [Play Store or F-Droid]

Depau, 2020-05-17:
@emayljames Hi, try keeping your phone still on a table while it's working, do not move it. If you have a better USB adapter, try it.

Depau, 2020-05-21:
If it works like that, let me know. Otherwise I need you to send me a logcat.

Error -7 is a timeout anyways, which means the error is due to external causes: http://libusb.sourceforge.net/api-1.0/group__libusb__misc.html#gab2323aa0f04bc22038e7e1740b2f29ef

If you can send me a logcat, and maybe also a `lsusb -v` output from a Linux machine, I'll see if I can implement some quirks if your USB drive consistently shows this issue.

I'll close this issue, if you still need help I can reopen it.

closed 2020-05-21 by Depau