etchdroid/etchdroid #54: [Bug] Fails to successfully write anything over 1GB

Gen0, 2020-06-04:
**Describe the bug**
Few days of constant, pretty intense testing. Multiple Linux and *ix distros tested on a new branded USB-C native pen drive. 
Distros tested
PuppyLinux - fails
Debian Dog - works
Kali 32 lite - fails
Kali 64 heavy - fails
OpenDarwin - fails
Tinycore - works
MacOS Mojave dmg created via --createinstallmedia to try and use DMG functionality. - can't use because "drive selection" window only gives me external partitions and all give errors on pressing. 

Basically all failures come in the form of disconnect errors in the mid 30% range. Whenever they didn't, they were consistently on big files like squashfs bundles. 

**To Reproduce**
Steps to reproduce the behavior:
1. Begin ISO burn
2. Wait a few minutes
3. Write failed at block 3964928, 2.0 GB written. Error: java.io.IOException: Could not claim interface, errno: 0 null

**Smartphone (please complete the following information):**
 - Device: Huawei P30 Pro
 - OS: 10.0.0.195 (EMUI so still Android with Google) 
 - Installed from: Play Store

**Additional context**
Interface at the port gets super hot 🔥

This is a USB-C pendrive and pushes 20MB/s. I don't have a USB-C to female USB so I cannot test other pendrives.

This could probably be fixed by giving transfer rate limiting options, and allowing ISO write-in-progress preservation so that writes can be finished if the device disconnects due to heat.

The DMG menu that asks you to pick a drive (for conversion to ISO I assume)  tells me it doesn't have the right to read any of the listed options. I renamed the DMGs to ISO and they appear to burn jsut fine but fail after 2GB like the rest of the bigger ones.

Depau, 2020-06-04:
Alright, thanks for your report. I'm sorry but add yourself to #10, this is just a different iteration of the same bug.

Also check this out: https://github.com/magnusja/libaums/issues/253

What's going on is that due to some change that neither I nor @magnusja have yet figured out, starting from Android 9, this happens. I'd like to tell you more but we've been fighting against this for more than 1 year. The only (applied) mitigation we've found is switching to libusb for the implementation, though as you've noticed it still occurs.

We suspect this relates to this issue on QEMU with an emulated USB drive https://github.com/magnusja/libaums/issues/209, and I'm quite convinced that if we managed to make it run on QEMU it will also fix the issue on real USB drives.

But as I said in the other issue, I'm quite busy right now to put the amount of effort that is required.

I'll close this, feel free to comment on #10 or even better https://github.com/magnusja/libaums/issues/253

Depau, 2020-06-04:
Additionally:

> Interface at the port gets super hot 🔥

Well, not much I can do about it software-wise :)

> This could probably be fixed by giving transfer rate limiting options, and allowing ISO write-in-progress preservation so that writes can be finished if the device disconnects due to heat.

I don't think it disconnects due to heat. This is a veeeeery known issue, it happens on all sorts of devices, including my laptop on Android x86 and I can assure you USB drives don't pop off my laptop's sturdy and brand new USB ports due to heat :)

The quality of the OTG adapter might affect this, but it's likely a software bug. Also cause it doesn't happen at all on Android < 8, unless you actually wiggle the drive.

> The DMG menu that asks you to pick a drive (for conversion to ISO I assume) tells me it doesn't have the right to read any of the listed options. I renamed the DMGs to ISO and they appear to burn jsut fine but fail after 2GB like the rest of the bigger ones. This is a separate bug entirely

I'm confused, this might need a separate issue. Could you explain it better?

The DMG menu converts and flashes on the fly a DMG image to a USB drive. Due to my poor choice of file browsers, and since it needs to directly access the file (so it doesn't work with Android Storage Access Framework), the file browser won't show anything unless you have a DMG image on your storage somewhere.

`dmg2img` is used under the hood.

closed 2020-06-04 by Depau

Gen0, 2020-06-04:
[image]

This notification is shown regardless of option chosen. Note that there is no USB inserted in this screenshot, I assume those 2 3GB are system/data/RAM partitions. 
I can move this to a new bug report if convenient. 😊

Depau, 2020-06-04:
> I can move this to a new bug report if convenient.
Done :)

> Fantastic reply rate btw

https://t.me/GitHubBot ;)