etchdroid/etchdroid #87: Can the app write DBAN?

07416, 2020-07-29:
I downloaded the ISO from the website https://sourceforge.net/projects/dban/files/dban/dban-2.3.0/dban-2.3.0_i586.iso/download, wrote with the app but my PC wasn't able to boot from the drive - using a desktop burner DBAN was able to start.

Depau, 2020-07-29:
@07416 In general, if the installation instruction say you can use dd or etcher to write it to a USB drive, it will work with etchdroid.

I can't check right now but i read from the filename that dban is for i586, so I'm guessing it's a retro Linux distro. Many older distributions only work on actual optical medium, you may have some luck with unetbootin on a computer with those.

In general support for this kind of images is a long term goal but a lot of technical issues need to be tackled for them to work

07416, 2020-07-30:
[DBAN](https://en.m.wikipedia.org/wiki/Darik%27s_Boot_and_Nuke) is a program to wipe HDDs, but is indeed based on Linux.

Depau, 2020-08-09:
Hi,
Sorry for the delayed response. I checked the DBAN image and it's a CD-only image, so EtchDroid won't work., at least not just yet

You may have better luck with unetbootin on a computer.

closed 2020-08-09 by Depau

07416, 2020-08-10:
"CD-only"? I booted DBAN from an USB just recently.

But sure, this is getting off-topic...

Depau, 2020-08-10:
@07416 You probably used Rufus