etchdroid/etchdroid #88: [Bug]

Forever4sleep, 2020-08-09:
Hello, I've been trying to write an Ubuntu iso to the flash drive, but it starts to hang on, about, 5-6% of writing, after some time it gets a notification of fail, here's the error log message: 

"Write failed at block 268288, 137.4 MB written. Error: java.io.IOException: libusb returned -1 in control transfer"

I've already tried to reconnect the device, but it wouldn't work. 

My phone info:
- Device: Honor 7A
- Model: DUA-L22
- Android version: 8.1.0

Depau, 2020-08-09:
Hi,
Can you try with a different USB drive if you have one and let me know?

This should hopefully by fixed once https://github.com/magnusja/libaums/pull/260 ends up in a release, which will be merged in the next release of the app.

Forever4sleep, 2020-08-09:
Unfortunately, I have no another device.

Depau, 2020-08-09:
@Forever4sleep You'll have to wait for the next etchdroid release then :/

Forever4sleep, 2020-08-09:
> @Forever4sleep You'll have to wait for the next etchdroid release then :/

Ok, thank you, how long it takes to the new upd?

Depau, 2020-08-09:
It won't happen before a new release of libaums and I don't know when it's due.

Forever4sleep, 2020-08-09:
> It won't happen before a new release of libaums and I don't know when it's due.

Hey, I've spotted a bug, no, it's even a future of your app; so, I decided to try to write the iso again, before this I had done these actions:

- Renamed the iso name to "bubuntu"
- Replaced it to the new created folder on the SD card 

Well, when it got to 6%, I for some reason decided to open your app, then, I spotted that the writing started to continue, and it'd been continuous before my phone got dead, but I guess that had my phone not dead, it would've been done. (IDK, maybe it's not because of the opened app, but because the steps I've done and described above)

Depau, 2020-08-09:
So you tried to open the app while it was still writing?

Yes it will let you try to write it while it's already writing. No I won't fix it takes too much effort. Also it's quite stupid to try doing it and thinking good things will happen (see gif below), but it won't do anything bad anyway. It will just screw up both writes and invalid data will end up in the USB drive, and you'll have to rewrite it from scratch.

When I'll implement the progress activity it should be "fixed" since trying to reopen the app will show the progress screen.

[image]

Forever4sleep, 2020-08-09:
@Depau Nevertheless, it worked perfectly and I'm writing this via PC, thank for the app :D

Depau, 2020-08-09:
👌

closed 2020-08-09 by Depau