etchdroid/etchdroid #89: Usb not showing up

sebin-gg, 2020-08-13:
Hi after using etch droid when I plug in my usb drive it doesn't show up I mean it doesn't even show the unsupported usb drive notification please do something about it I really need that pendrive I am using a vivo phone

Depau, 2020-08-13:
@sebin-gg Have you tried to reboot your phone?

sebin-gg, 2020-08-13:
Yes

Depau, 2020-08-13:
@sebin-gg That's pretty weird.

Is your phone rooted or do you have another device that is rooted? In alternative, do you have a Linux computer? As a third alternative, any other computer can work, but I'd rather have you do some troubleshooting from a (rooted) Linux phone or computer.

In any case, can you please provide a logcat taken after plugging and unplugging after a few seconds the USB device? No need to open the app

Depau, 2020-08-15:
Okay, then connect the USB drive to the Linux/Android device, disconnect it after a few seconds, run `dmesg > kernel.log` (as root if on Android) and send the log.

sebin-gg, 2020-08-16:
I don't have   have rooted android device or a pc other than my laptop which i was trying to flash an os onto

Depau, 2020-08-16:
@sebin-gg IDK how to help you then :/ No one you can borrow a different device / USB drive from?

sebin-gg, 2020-08-16:
Let me try

Depau, 2020-08-16:
@sebin-gg It might also be an issue with the OTG adapter, those things break more often than you'd expect, I always buy a few extra to avoid this kind of issues.

sebin-gg, 2020-08-16:
And I found something interesting when I downloaded etch droid from f droid on my old samsung 5year old galaxy 5or3 it crashed as soons as it accessed the raw fil

Depau, 2020-08-16:
@sebin-gg What Android version?

sebin-gg, 2020-08-16:
It's android version 5 or 4.4 I don't have it wright now

Depau, 2020-08-16:
@sebin-gg If it's android 4.4 (and I think that's the case considering the issue you mentioned) you're using a old version of the app, support for android 4.4 was dropped over a year ago.

sebin-gg, 2020-08-16:
But now iam using an oppo y11 andoid 9 to flash the os and the pendrive is not showing up😔

Depau, 2020-08-16:
@sebin-gg do you have a USB mouse/keyboard to try to see if the adapter is broken?

sebin-gg, 2020-08-16:
I have tried what you said and found my otg cable is broken

Depau, 2020-09-01:
Alright, closing this then :)

closed 2020-09-01 by Depau