etchdroid/etchdroid #90: [Bug] Android proihibits other applications from accessing the drive while the formatting happens

Wilker-a, 2020-08-17:
[image]

the application is prevented from writing to the sd card when Android stops recognizing the file system at some point and requires the drive to be formatted.

Depau, 2020-08-17:
I'm not completely sure what you mean.

Please provide a step by step guide to reproduce the issue.

Wilker-a, 2020-08-17:
i'm sorry, right after i posted this the program showed me a dialog talking about the issue ^^' it seems that this is already acknowledged.

Android basically says that it does not recognize the drive during process of installing and revokes access from everything else trying to access it.

that said, rebooting my device did work but now neither Android nor my pc is recognizing the usb like what was installed before (i overwrote Debian+GNOME installed using Rufus with an image with KDE). i'll look in to it to see if the issue isn't the image file (the page from Debian had like 4 torrent files of the same version available and it was confusing)

Wilker-a, 2020-08-17:
i am closing this issue since it turned to be redundant and probably duplicated. thanks for everything \^\^

closed 2020-08-17 by Wilker-a

Depau, 2020-08-17:
@Wilker-a It is normal that Android says it's not recognized. If the computer doesn't boot you should read the motherboard/bios manual to see how to configure it to boot of of the USB drive

Wilker-a, 2020-08-17:
i don't have a manual, it is really old too (i doubt that i can find one for my motherboard model) \^\^'

but the issue was that the downloaded ISO only had packages but no boot