etchdroid/etchdroid #92: [Bug]

scshiv29-dev, 2020-09-06:
**Describe the bug**
I tried making a bootable drive of Ubuntu 20.04 from app , it started and said that writing completed and yet the laptop isn't detecting it.. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. See error

**Expected behavior**
.
A clear and concise description of what you expected to happen
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Smartphone (please complete the following information):**
 - Device: [e.g. Xiaomi K20PRO]
 - OS: [e.g. Android 10]
 - App version [e.g. 1.4, you can find it in the About screen or in system Settings]
 - Installed from: [Play Store,]

**Additional context**
Add any other context about the problem here.

Depau, 2020-09-06:
@scshiv29-dev Hi, you should follow the installation guide fully. If the write completed successfully there is no issue. https://ubuntu.com/tutorials/install-ubuntu-desktop

scshiv29-dev, 2020-09-06:
That's what i think but , it's not detecting in the BIOS this is first time happening.. i tried with elementary os and it was working

Depau, 2020-09-06:
@scshiv29-dev Try writing it again, if the app does not report any error, the problem is not in the app. You should check if the ISO you downloaded is correct, that it is not corrupted, that your USB drive works correctly and that you're following the correct procedure for your computer to boot off of USB.

scshiv29-dev, 2020-09-06:
I guess the problem is with the iso thanks

closed 2020-09-11 by Depau