etchdroid/etchdroid #93: An error

TimeGaming2030, 2020-09-15:
I was just quilcky flashing ubuntu to a usb and this is the error it have me "Write failed at block 1914880, 980.4 MB written. Error: java.io.IOException: libusb returned -4 in control transfer" thonit workds after i re formatted the usb and restarted my device so yeh.

Depau, 2020-09-15:
Hi,
You should look for a better USB-OTG dongle:

[image]

https://vovkos.github.io/doxyrest/samples/libusb/enum_libusb_error.html#doxid-group-libusb-misc-1ggab2323aa0f04bc22038e7e1740b2f29efa01b43115484f435627be755d5dc9509b

TimeGaming2030, 2020-09-15:
K thanks for leting me know

Depau, 2020-09-15:
Alright, I'll close this.

closed 2020-09-15 by Depau