etchdroid/etchdroid #110: Unplugged bug!!!

Gamer-one-o-one, 2020-12-12:
My USB device is plugged in. But after its done 2%,  an error pops up and says my usb might unplugged during the process. Its says its an unknown error or bug i dont know. I am using it on android 10 for an windows recovery.

Depau, 2020-12-12:
@Gamer-one-o-one Hi,

1) Windows recovery will not work. Did your read the project description?

2) I also need the error message

closed 2020-12-16 by Depau

Gamer-one-o-one, 2020-12-16:
Hello Developers,

Thank you for your reply

  1.If I use this software to create a usb bootable drive than shouldn't it
work as a recovery drive also. Sorry if I am wrong. I am not pro at this
😅(I am not using any third party software, using the original win os image
)

 2. Here is the error. It shows only after 2% is done.

Plzz help.

Thank You

Gamer-one-o-one, 2020-12-16:
I am sorry.
Sorry man.
I Didn't read the project description.
>Windows OS is not supported 😶.
Sorry for wasting your time.

Sincerely,
Dumb user