etchdroid/etchdroid #115: Migrate Kotlin synthetics to ViewBinding

TacoTheDank, 2021-01-11:
https://developer.android.com/topic/libraries/view-binding/migration

Depau, 2021-01-11:
Hi,
I don't actually use that, I already checked it and it works by simply removing `apply plugin: `kotlin-android-extensions``. In fact, it should be gone from master unless I forgot to push.

ViewBinding definitely needs to be switched to from the current spaghetti UI. However, I have no time for that at the moment :/

TacoTheDank, 2021-01-11:
@Depau I still see usages [here](https://github.com/EtchDroid/EtchDroid/search?q=synthetic), so perhaps you forgot to push :)

Depau, 2021-01-12:
Oh okay I didn't forget to push it, I didn't fix it :D

Well I guess it will stay like that for some time until I have some time or someone fixes it, I currently have higher priority stuff on my todo list :/

TacoTheDank, 2023-04-12:
The Compose rewrite fixed this, it seems.

closed 2023-04-12 by TacoTheDank