etchdroid/etchdroid #117: Does not work on Samsung S20 / OneUI 3.0 / Android 11

alexgranford, 2021-01-30:
Fails on USB destination selection.
Phone: Samsung S20 with OneUI 3.0 and Android 11
USB SD Reader: tried with Anker USB C 3.0, Kingston USB A 3.0 via USB C adaptor
App version 1.5 installed from Play Store

**To Reproduce**
Steps to reproduce the behavior:
1. Write raw image or ISO
2. File chosen (Raspberry Pi RaspOS .img file)
2. Select USB drive dialog: 
2.1 shows "Realtek USB3.0 Card Reader" /dev/bus/usb/002/002
2.2 click on that destination
2.3 Ask to allow access? OK
2.4 Immediately gives system message "Restart your phone"
2.5 App gets frozen, in 5 seconds system alert with "EtchDroid isn't responding"

Depau, 2021-01-30:
>  2.4 Immediately gives system message "Restart your phone"

What does this message say exactly?

Are you able to collect a logcat and send it? I ultimately need it to check your issue, I don't have any Samsung devices.
You can find guides online for your operating system, a computer is required.

alexgranford, 2021-01-30:
Message popup is a system notification under "System UI", saying

Header: Restart your phone
Message: A USB Storage device was removed unsafely. To prevent your phone from restarting unexpexpectedly, save unsaved data and restart now
Action: "Restart" button.

Your app does not continue into showing next step and just hangs.

I'm not able to provide logcat, not rooted and don't want to mess around with adb at this stage.

alexgranford, 2021-01-30:
PS. I have also tried Raspi Card Imager app.

It also fails with this error: java.io.IOException: Cannot run program "su": error=2, No such file or directory.

Not sure if that is a common issue or just theirs. Also not sure why would it need to run su on non-rooted device.

Depau, 2021-01-30:
The other app fails because it's assuming your device is rooted, for some reason.

EtchDroid does not use root even if the device is rooted. This issue is more likely to be related to the USB API, since it's tricky to test and the emulator doesn't support it, bugs slip in very easily.

>  Message: A USB Storage device was removed unsafely. To prevent your phone from restarting unexpexpectedly, save unsaved data and restart now

I see, so it's Samsung crap. What happens if *before using EtchDroid*, while the app is fully closed, you eject the USB drive from the notification panel? I think there should be a button to do so.

alexgranford, 2021-01-30:
When I plug a new USB storage, notification panel adds this:

> Settings
> USB storage added
> Buttons: Open, Unmount

When I press Unmount, it shows "VOLUME1 unmounted"

Depau, 2021-01-30:
> When I plug a new USB storage, notification panel adds this:
> 
> > Settings
> > USB storage added
> > Buttons: Open, Unmount
> 
> When I press Unmount, it shows "VOLUME1 unmounted"

Yes. try pressing unmount, then run EtchDroid. I think it will work.

alexgranford, 2021-01-30:
> Yes. try pressing unmount, then run EtchDroid. I think it will work.

So yes we have some progress here as there is no "Restart your phone" message any more.

But - it still hangs straight when I press the USB storage, it does not move to the next screen.

Depau, 2021-01-30:
Fully close the app, if needed find it in settings and force stop it or reboot the phone. Then try again, but unmount the drive first.

The app hangs when the USB permission has been granted but something bad happened after that, and the only way I could find to recover from it is fully kill it and restart it. It needs a fresh USB permission.

alexgranford, 2021-01-30:
Didn't work after both forced app kill from settings, and after full reboot and cleaning cache / data in app settings.

Steps:
[no app running]
plug in -> unmount -> start app and follow screens.

Depau, 2021-01-30:
Then I'm not sure what is going on, I need a logcat. Alternatively you can obtain a bug report, but please don't post it here as it may contain sensitive info, please email it to me.

https://developer.android.com/studio/debug/bug-report

alexgranford, 2021-01-30:
[image]

alexgranford, 2021-01-30:
Does not seem to be too helpful imo....

Depau, 2021-01-30:
Yeah, it isn't very hepful. IDK how to help you though, since

- I don't have any Samsung devices
- I don't have any Android 11 devices (and in fact the app has never been tested with Android 11)
- It can't be tested on the official emulator since it doesn't support emulating or passing through USB devices

I'm about to order a Pixel 5, however, so if the issue is caused by Android 11 and not by Samsung's additions I'll find out.

bloodycorps3, 2021-02-21:
Same problem     someone   fix it ?

ricarva, 2021-04-18:
Just to chime in, this happens also on a OnePlus 6T running Android 10: app hangs after USB device select.

Would a logcat still help to solve this?

ricarva, 2021-04-18:
@Depau, hi

Sorry to disagree: as I stated above, I'm getting the exact same problem on a OnePlus device, not Samsung.

And on the Samsung device I do have (Tab S7), ejecting the device before attempting to flash only gets rid of the warning notifications; the flashing process always hangs at the same spot, regardless, and on both the Samsung and OnePlus devices.

Depau, 2021-04-18:
@ricarva Yes but it's probably the same issue: manufacturer customizations. It does not happen on plain Android 11, I have a Pixel 5 and I can't reproduce the issue. No need to send me a logcat, when the app freezes before starting it means the USB drive is in use by something else and unless you have other apps fighting over it, it's probably in use by the system. Try to find a setting to eject it.

ricarva, 2021-04-18:
@Depau 

I've looked on both devices,  have ejected my SD Card/Reader, and still no luck on either the Samsung or OnePlus.

It's quite strange that two devices from two completely unrelated brands would have the same customization.

Depau, 2021-04-18:
> It's quite strange that two devices from two completely unrelated brands would have the same customization.

It's not.

Anyway I do not have a OnePlus device affected by the issue and neither a Samsung so I can't help you :/

The code is here, feel free to send pull requests.

lypanov, 2021-04-27:
Same issue with Samsung Tab S6 w OneUi 3.1 and with Huawei P40 Pro+ w Android 10.

I'll look into running adb/logcat whenever I find the time. Glad to find this issue at least, thought I'd just purchased a bad USB drive for no food reason...

Depau, 2021-04-27:
I was able to borrow a Samsung Galaxy Tab A w/ OneUI 2.1 and I can't reproduce the exact issue but it does crash.

The bad news is that for some reason it won't let me connect to ADB over Wi-Fi so I can't see why it's crashing. On top of that, I'll have to return it.

So unless I can figure it out by tonight, I'll need a pull request from you guys.

If you guys can confirm that opening the storage settings and clicking "Eject" works, and you can also tell me the name of the activity that displays this option, I can add a shortcut to it in the main screen when a samsung device is detected.

ricarva, 2021-04-27:
@Depau, as previously stated, on my devices (Samsung and OnePlus) the "eject" procedure does not enable successful operation of your app.

Cheers

lypanov, 2021-04-27:
Ditto for Samsung + Huawei. Ejecting doesn't help, just avoids the eject warning on the Samsung. After that just getting a blank page followed by a "... not responding" popup.

Depau, 2021-04-27:
Since this issue is actually in the underlying USB drive access library libaums, I moved the issue here. The discussion can continue on the other issue.

I'll lock this, but I'll leave this open and pin it so that people see it.