etchdroid/etchdroid #118: Have an option to reformat a flash drive completely

lmore377, 2021-02-01:
**Is your feature request related to a problem? Please describe.**
With some disks that have multiple partitions (take a disk with raspbian as an example) android will sometimes only see the first partition so when you reformat it you'll only get the first partition back and sometimes it'll show all the partitions but it won't let you reformat the whole disk as one.

**Describe the solution you'd like**
An option that will either format the flash drive or erase/corrupt the partition table so that way android can handle reformatting it

**Describe alternatives you've considered**
Selecting a random file (text file, an actual image, etc.) and writing it to the flash drive messes up the partition table so android completely reformats the drive and achieves the desired result.

Depau, 2021-02-01:
Hi, good workaround :)

I'll keep this open since it's a useful feature that I'd eventually like to have, but at the moment I don't have time to work on it :/

maxmod, 2021-05-19:
Use this software working https://bit.ly/3oBO5JO