etchdroid/etchdroid #121: [Bug]

The-C-programmer, 2021-02-12:
**Describe the bug**
I am trying to flash an image of Linux on my USB flash drive. EtchDroid says flash complete, but when I insert the USB flash drive it says "This is not a bootable floppy. Please insert a bootable floppy and press any key to try again..."

Steps to reproduce the behavior:
1. Go to Play store and download the latest version of EtchDroid.
2. Download any image of Linux from the web and flash it with EtchDroid as usual.
3. Boot your computer, with the USB flash drive inserted and from the boot menu select your USB drive.
4. See error

**Expected behavior**
I expected to see a screen with the installation menu.

Screenshots
[image]
[image]

**Smartphone (please complete the following information):**
 - Device: ACRUX, (tablet, from a educational plan from my country)
 - OS:  Android 9 Pie
 - App version (1.5)
 - Installed from: [Play Store]
(Note: I have flashed an image of Ubuntu 15.10 with no problems)

Depau, 2021-02-12:
> Download any image of Linux from the web and flash it with EtchDroid as usual.

What distribution did you download?

The-C-programmer, 2021-02-12:
I downloaded Arch Linux 32 from its official site. It is a continuation og Arch Linux for 32 bit PCs

Depau, 2021-02-12:
Does the app show an error? If it doesn't, then the problem is either with the image you're flashing or with your computer, since older machines with ≤i686 CPUs usually have more issues booting from USB.

It is also possible that AL32 does not support booting off USB. You can try using unetbootin to write the image, it uses a different technique for writing the image and uses a bootloader made specifically for USB booting. EtchDroid does not support unetbootin-style flashing.

If that doesn't work you can try booting it with something like Plop Boot Manager (which you need to burn onto an actual CD or floppy disk), test different USB modes from its config and see if it boots.

I'm 99.99999999% sure your issue is not caused by EtchDroid, so I'm going to close this.

closed 2021-02-12 by Depau