etchdroid/etchdroid #130: [Bug] USB capacity is not -512.

Zel9278, 2021-04-19:
[image]
I accidentally deleted the grub on my PC and I'm trying to recover it, but I only have Android and USB at hand and I'm trying to burn it using this app.
The OS is Android 11.
The device is a Galaxy S20+.

Depau, 2021-04-19:
@Zel9278 ... get a bigger USB drive?

Depau, 2021-04-19:
@Zel9278 Mmh it's not supported at the moment. Big disks use a different API. Use a smaller one.

Zel9278, 2021-04-19:
oh.... ok

Depau, 2021-04-19:
Btw, since it's a Samsung, you may also want to give a look at this https://github.com/EtchDroid/EtchDroid/issues/117

Recent Samsung roms require you to eject the drive from settings beforehand.

Zel9278, 2021-04-19:
> Btw, since it's a Samsung, you may also want to give a look at this #117
> 
> Recent Samsung roms require you to eject the drive from settings beforehand.

I tried, but the drive keeps reconnecting.

Depau, 2021-04-27:
Since this issue is actually in the underlying USB drive access library libaums, I moved the issue here. The discussion can continue on the other issue.

I'll close this issue.

Depau, 2021-04-27:
By the way, the next release will show "Unsupported" + an explanation message instead of showing a negative size. Thanks for reporting!

closed 2021-04-27 by Depau