etchdroid/etchdroid #135: Revoking permission at SecondaryChooserFragment crashes app

aper-project, 2021-05-07:
Hi, there, I found **EtchDroid v1.5** (downloaded from F-Droid) crashes when a permission is revoked.

I think it is related to the dangerous permission `android.permission.WRITE_EXTERNAL_STORAGE` (or  `android.permission.READ_EXTERNAL_STORAGE`) and is triggered during the following call chain:
```
com.codekidlabs.storagechooser.fragments.SecondaryChooserFragment$7.onClick(android.view.View)void
 com.codekidlabs.storagechooser.fragments.SecondaryChooserFragment.access$1700(com.codekidlabs.storagechooser.fragments.SecondaryChooserFragment)void
  com.codekidlabs.storagechooser.fragments.SecondaryChooserFragment.performBackAction()void
   com.codekidlabs.storagechooser.fragments.SecondaryChooserFragment.populateList(java.lang.String)void
    com.codekidlabs.storagechooser.fragments.SecondaryChooserFragment.setBundlePathOnUpdate()void
     android.os.Environment.getExternalStorageDirectory()java.io.File
```

This call chain does not involves either permission checking (via [checkSelfPermission](https://developer.android.com/reference/androidx/core/content/ContextCompat#checkSelfPermission(android.content.Context,%20java.lang.String))) nor permission request (via [requestPermissions](https://developer.android.com/reference/androidx/core/app/ActivityCompat#requestPermissions(android.app.Activity,%20java.lang.String[],%20int))) to the corresponding permission, which may produce unexpected consequence.

#### Reproduce crash

In fact, I found this lack of permission management causes a **crash** on my Samsung device, following the steps:

1. in the main activity, click "write apple DMG image" with **granting the storage permission**
2. select either storage directory (in the menu of "Choose Storage")
3. go to system setting and **revoke the storage permission**
4. go back to the app and it crashes

@Depau
Could you help me review this issue? If you need more information, I may make a video showing the crash. Very thanks!

Depau, 2021-05-07:
The app cannot write DMGs without the storage permission.

If you do not need to write DMGs, do not grant the permission.
If you need to write DMGs the storage permission is required.

Depau, 2021-05-07:
FWIW the permission grant is checked: https://github.com/EtchDroid/EtchDroid/blob/develop/app/src/main/java/eu/depau/etchdroid/ui/activities/ActivityBase.kt#L63-L77

closed 2021-05-07 by Depau

Depau, 2021-05-07:
I think what you're doing is reverting the storage permission while the app is running, I suppose your Samsung ROM still returns granted until you restart the app but then the system still denies access to the resources. Or something along these lines.

TBH this is a super edge case and I'm not going to spend precious time wondering "what if the user does takes this super weird path"