etchdroid/etchdroid #139: [Bug]

inet56, 2021-05-14:
**Describe the bug**
Write is failing with error like this,

Write failed at block 63488, 32.5 MB written. Error: java.lang.IllegalArgumentException

It says on notification that the device may have been unplugged, while it isn't.

**To Reproduce**
Steps to reproduce the behavior:
Perform usual procedure for iso file.

**Expected behavior**
Write to finish.

**Smartphone (please complete the following information):**
 - Device: Symphony i10+
 - OS: Android 10 Go
 - App version 1.5
 - Installed from: Play Store

**Additional context**
I am writing Fedora 34 Workstation.

Depau, 2021-05-14:
Are you able to take a logcat from the device using adb? Can you provide it?

inet56, 2021-05-14:
Sorry, I don't have a secondary machine to perform requested operation.

Depau, 2021-05-14:
You can create a bug report on the device itself, then use a random zip app to extract the log from there.

Anyway I just checked the google play console and I can't find any crashes that match your description.

Which might mean that most likely there's an issue with your USB adapter or USB drive, otherwise it would happen to a lot more people.

inet56, 2021-05-14:
> Which might mean that most likely there's an issue with your USB adapter or USB drive, otherwise it would happen to a lot more people

I can transfer files to the USB device (sdcard on a adapter) including the iso file.

Depau, 2021-05-14:
@inet56 Then I need the log, there's no way around it. You can google instructions on how to extract logs from bug reports without requiring an external device

inet56, 2021-05-14:
There is no 'Take a bug report' option on this device as told on https://developer.android.com/studio/debug/bug-report#bugreportdevice

The device reboot while a write was performing last two times.

closed 2021-05-14 by inet56

Depau, 2021-05-14:
Then there's nothing I can do to help you :/