etchdroid/etchdroid #143: [Bug]

Kunal-0403, 2021-06-09:
It doesn't flash the Linux distro in my hp pendrive.

It shows this message:-
Write failed at block 284672, 145.8 MB written. Error: java.io.IOException: libusb returned -7 in control transfer

**Screenshots**
If applicable, add screenshots to help explain your problem.

 - Device: Redmi 5A
 - OS: Android 8
 - App version: 1.5
 - Installed from: Play Store

path57, 2021-06-11:
Hi,
Getting the same error
Write failed at block 12288, 6.3 MB written. Error: java.lang.IllegalArgumentException
Doesn't always stop at the same amount of written data.
USB drives are fine and android formated.
OS android 10
Vankyo Z1 tablette

path57, 2021-06-11:
Nothing strange in the system

Kunal-0403, 2021-06-12:
Don't you have any suggestions? I'm struck with an unbootable PC and I don't have another one not I can borrow from someone during lockdown.

Depau, 2021-06-12:
>  Doesn't always stop at the same amount of written data.

Then it's a low quality USB adapter. Avoid moving it while it's writing. Or just get a better one.

I'm not aware of issues with redmis and Android 8.

Depau, 2021-06-12:
As a side note, I'm sick and tired of people who don't fill in the entire form when submitting issues. From now on I won't even read them and close them.

closed 2021-06-12 by Depau

Kunal-0403, 2021-06-12:
It's HP USB 2.0 Flash Drive v236W 32GB

path57, 2021-06-12:
Same here I would need another Windows or Linux PC to make the boot ant USB
stick

Patrick Hollinger

Depau, 2021-06-12:
I don't have any other suggestions. If it doesn't work it doesn't work :man_shrugging: