etchdroid/etchdroid #148: feature request: add compression formats support

ilyapashuk, 2021-07-22:
hello

please add an ability to directly, without unpacking, write images in formats like gz and xz with on the fly decompression.

it is not very usefull for usual iso's, because most of them are compressed internally, by technologies like squashfs, but this can be very usefull for raspberry pi images.

s2l1, 2021-07-22:
this would be wonderful for flashing a manjaro arm .img.xz on micro SD cards 
+1

Depau, 2021-07-22:
I don't have time to work on it.

If someone can send a pull request I will review it.

s2l1, 2021-09-10:
> I don't have time to work on it.
> 
> If someone can send a pull request I will review it.

Would love to offer a bounty payable in Bitcoin if you are open to the idea! We are working on a plug n' play setup that this feature would really be handy for.

Depau, 2021-09-10:
@s2l1 I'm not interested. And I'm especially not interested in crypto payments. If you want you can add a bounty to bountysource so that people can claim it.