etchdroid/etchdroid #157: Fails every time

SerbianSlav2, 2021-12-21:
Write failed at block 296960, 152,0 MB written. Error: java.io.IOException: libusb returned -7 in control transfer

ISO:https://spins.fedoraproject.org/kde/download/kde-download-splash?file=https://download.fedoraproject.org/pub/fedora/linux/releases/35/Spins/x86_64/iso/Fedora-KDE-Live-x86_64-35-1.2.iso

Device:Nokia 5.1

OS: Android 10

RoverBot-Dev, 2022-01-12:
Thats interesting. I get the same error message and i am on android 10. Sometimes i don't get the error but usb doesn't boot on PC

Ryanmustofa23, 2024-02-23:
Bisakah saya mencoba aplikasi terbaru, saya mencoba menjadi tester

depau, 2025-05-02:
The latest release should allow you to reconnect the USB drive if writing fails.

closed 2025-05-02 by depau