etchdroid/etchdroid #158: [Bug] Please fill in the title

davirec, 2022-03-24:
In the double screen rotation test to detect data loss, an application failure was identified: lost fragment and lost fragment text.

- AboutActivity:lost fragment
<div id="ladoalado" style="display:flex;justify-content:center;">

<figure id="container">Before and after double screen rotation<br>
[image]
</figure>

<figure id="container">
[image]
</figure>

</div> 

- StartActivity:lost fragment text
<div id="ladoalado" style="display:flex;justify-content:center;">

<figure id="container">Before and after double screen rotation<br>
[image]
</figure>

<figure id="container">
[image]
</figure>

</div> 

A data loss problem occurs when data is accidentally deleted or state variables are accidentally assigned with default or initial values. This issue is related to the activity's lifecycle (1). When a screen rotation occurs, the activity is destroyed and resumed. Data loss will happen if the developer does not save the variables before the destruction and restore them after creation. To avoid this, the developers have to implement both the logic necessary to save the activity state in the onSaveInstanceState() callback method and the logic to resume its state in the onRestoreInstanceState() callback method.
Another way to avoid data loss problems is to use the view model, designed to store and manage UI-related data in a lifecycle-conscious way (2).

Note:

- Double screen rotation: It is executed twice to reach a state that should be exactly the same as the screen before if no data loss occurred.
- A data loss problem occurs when data is accidentally deleted or state variables are accidentally assigned with default or initial values.

1. https://developer.android.com/guide/components/activities/activity-lifecycle 
2. https://developer.android.com/topic/libraries/architecture/viewmodel

Depau, 2022-03-24:
I hope you're not doing this for free.

[image]

Anyway I'm definitely not rushing to fix this. If anyone wants to fix the lifecycle, send a PR.

davirec, 2022-03-24:
Hi Davide Depau, 
I'm doing scientific research in software engineering involving robotics and automated testing tools. We use DLD (Data Loss Detector) to generate computerized tests to find data loss on Android devices.

Depau, 2022-03-25:
Good luck on your research!

depau, 2025-05-02:
Fixed with Jetpack Compose rewrite

closed 2025-05-02 by depau