etchdroid/etchdroid #185: [Bug] ANR after choosing image from internal storage

amenekowo, 2023-06-02:
## Description

After choosing image from internal storage, it goes black screen and Android informed me this app is not responding.

## To reproduce

1. Click "Write raw image"
2. Select "test.img"
3. Choose usb reader
4. Goes anr

How many times have you tried? (type a number greater than 1):
Many, at least 10

## Expected behavior

Write image to my sd card

## Screenshots or logs

[image]

## Smartphone information

 - Device type (smartphone, tablet, laptop, etc): smartphone 
 - Device make (i.e. Samsung, OnePlus...): Xiaomi
 - Android version (i.e. Android 10): Android 13
 - ONLY IF YOU'RE USING A CUSTOM ROM, SUCH AS LINEAGE OS:
   - ROM name (i.e. LineageOS, PixelExperience...): Evolution X
   - ROM version: 7.9
   - ROM origin (official or unofficial?) Official
   - ONLY IF UNOFFICIAL ROM - LINK TO DOWNLOAD (XDA post, blog post, etc):
 - ONLY IF YOU'RE USING THE ORIGINAL OS INSTALLED BY THE MANUFACTURER:
   - ROM name (i.e. Samsung OneUI, Xiaomi MIUI): 
   - ROM version (i.e. OneUI 3.0):

## App information

- App version (find it in the About screen inside the app): 1.5
- Installed from (Play Store, F-Droid, GitHub or self-built): Play Store

## USB drive information

- USB drive capacity (i.e. 8 GB): 
- All information about the drive, AS SHOWN IN THE APP:
[image]

## USB port info

- Does the USB drive plug directly into your device without any adapters? (Yes/No): 

- If using a USB adapter, dock or hub:
  - USB port type ON THE DEVICE (put an "X" inside `[ ]`, like this: `[X]`):
    - [X] USB type C
    - [ ] Micro USB
    - [ ] USB type A (those commonly found on non-Apple laptops and desktops)
  - USB port type ON THE USB DRIVE (put an "X" inside `[ ]`, like this: `[X]`):
    - [X] USB type C
    - [ ] Micro USB
    - [ ] USB type A (those commonly found on non-Apple laptops and desktops)
  - Adapter make and/or where you bought it (Amazon, AliExpress, local store, bundled with phone...):
Huawei card reader Type-C port
  - Adapter type (put an "X" inside `[ ]`, like this: `[X]`):
    - [ ] Compact, solid microUSB/Type-C to USB-A OTG adapter
    - [ ] Flexible microUSB/Type-C to USB-A adapter
    - [ ] USB-C dock with > 1 USB port (usually advertised for laptops)
    - [ ] Other type of USB hub with > 1 port

## ISO or image file information

- Image file format (ISO, DMG, IMG): img
- Image file size AS SHOWN BY THE APP: 5.8G
- Linux distribution name and version (if it applies, i.e. Ubuntu 20.04): Raspbian buster
- Is this image for a Raspberry Pi or similar? (Yes/No): Yes

## Additional info

Tried downloading 1.5.1 from F-Droid, but still can't write (now I can access ready to write page) because "can't read usb device"
I clicked "Allow access to usb device" popped up before.
****
[image]

depau, 2023-12-25:
Hi, does the issue still happen with the latest version (pre-release) of the app?

amenekowo, 2023-12-26:
Unfortunately it doesn't work. (Latest beta version, 1.9 rc1)
[Here's the crash log.](https://paste.evolution-x.org/b837Sv)
In my opinion, maybe it's because some incompatible things here (I know little about android dev so I don't sure if it's right)
And the crash screen:
[image]

depau, 2024-01-07:
Hi, sorry about the delay, I've been busy lately.

Would you be able to send me the logs?

Launch the app, reproduce the issue then save the logs with something like [this app](https://play.google.com/store/apps/details?id=com.dp.logcatapp&hl=en&gl=US), or by running `adb logcat -d > logcat.txt` on your computer with your phone connected via adb.

Then I suggest you email them to me instead of attaching them to this issue since they might contain personal info; my email address is in my profile.

I'll take a look at them as soon as I have some free time (it might take a while).

depau, 2024-01-07:
Sorry I saw you sent the stack trace already, my bad. You can disregard my previous message.

amenekowo, 2024-01-09:
That's fine :)
If you need more information please let me know.