etchdroid/etchdroid #187: [Bug] Please fill in the title

Syntax8, 2023-07-28:
## Description

## To reproduce

How many times have you tried? (type a number greater than 1):

## Expected behavior

## Screenshots or logs

## Smartphone information

 - Device type (smartphone, tablet, laptop, etc): 
 - Device make (i.e. Samsung, OnePlus...): 
 - Android version (i.e. Android 10):
 - ONLY IF YOU'RE USING A CUSTOM ROM, SUCH AS LINEAGE OS:
   - ROM name (i.e. LineageOS, PixelExperience...):
   - ROM version:
   - ROM origin (official or unofficial?)
   - ONLY IF UNOFFICIAL ROM - LINK TO DOWNLOAD (XDA post, blog post, etc):
 - ONLY IF YOU'RE USING THE ORIGINAL OS INSTALLED BY THE MANUFACTURER:
   - ROM name (i.e. Samsung OneUI, Xiaomi MIUI): 
   - ROM version (i.e. OneUI 3.0):

## App information

- App version (find it in the About screen inside the app):
- Installed from (Play Store, F-Droid, GitHub or self-built): 

## USB drive information

- USB drive capacity (i.e. 8 GB): 
- All information about the drive, AS SHOWN IN THE APP:

## USB port info

- Does the USB drive plug directly into your device without any adapters? (Yes/No): 

- If using a USB adapter, dock or hub:
  - USB port type ON THE DEVICE (put an "X" inside `[ ]`, like this: `[X]`):
    - [ ] USB type C
    - [ ] Micro USB
    - [ ] USB type A (those commonly found on non-Apple laptops and desktops)
  - USB port type ON THE USB DRIVE (put an "X" inside `[ ]`, like this: `[X]`):
    - [ ] USB type C
    - [ ] Micro USB
    - [ ] USB type A (those commonly found on non-Apple laptops and desktops)
  - Adapter make and/or where you bought it (Amazon, AliExpress, local store, bundled with phone...):
  - Adapter type (put an "X" inside `[ ]`, like this: `[X]`):
    - [ ] Compact, solid microUSB/Type-C to USB-A OTG adapter
    - [ ] Flexible microUSB/Type-C to USB-A adapter
    - [ ] USB-C dock with > 1 USB port (usually advertised for laptops)
    - [ ] Other type of USB hub with > 1 port

## ISO or image file information

- Image file format (ISO, DMG, IMG): 
- Image file size AS SHOWN BY THE APP: 
- Linux distribution name and version (if it applies, i.e. Ubuntu 20.04): 
- Is this image for a Raspberry Pi or similar? (Yes/No): 

## Additional info

gaayu1234, 2023-08-30:
Type c OTG cable with pendrive

gaayu1234, 2023-08-30:
Jcbnnnf

closed 2023-12-25 by depau