etchdroid/etchdroid #190: [Bug] Attempt to get length of null array

lyk082401, 2023-09-08:
## Description

## To reproduce

How many times have you tried? (type a number greater than 1): 10+

## Expected behavior

## Screenshots or logs

[Device Brand]: Redmi
[Device Model]: 22021211RC
[Display]: TKQ1.220807.001 test-keys
[Android Version]: 13
[Android API Level]: 33
[System Locale]: zh_CN
[Process ID]: 17811
[CPU ABI]: arm64-v8a
[Package Name]: eu.depau.etchdroid
[Version Name]: 1.5.1
[Version Code]: 13
[Error Type]: JVM
[Crash Time]: 2023-09-09T05:10:41.050
[Stack Trace]:
java.lang.NullPointerException: Attempt to get length of null array
	at com.codekidlabs.storagechooser.fragments.ChooserDialogFragment.populateList(ChooserDialogFragment.java:262)
	at com.codekidlabs.storagechooser.fragments.ChooserDialogFragment.initListView(ChooserDialogFragment.java:115)
	at com.codekidlabs.storagechooser.fragments.ChooserDialogFragment.getLayout(ChooserDialogFragment.java:85)
	at com.codekidlabs.storagechooser.fragments.ChooserDialogFragment.onCreateDialog(ChooserDialogFragment.java:300)
	at android.app.DialogFragment.onGetLayoutInflater(DialogFragment.java:418)
	at android.app.Fragment.performGetLayoutInflater(Fragment.java:1322)
	at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1303)
	at android.app.FragmentManagerImpl.addAddedFragments(FragmentManager.java:2431)
	at android.app.FragmentManagerImpl.executeOpsTogether(FragmentManager.java:2210)
	at android.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManager.java:2166)
	at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:2067)
	at android.app.FragmentManagerImpl$1.run(FragmentManager.java:742)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:210)
	at android.os.Looper.loop(Looper.java:299)
	at android.app.ActivityThread.main(ActivityThread.java:8116)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:559)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)

## Smartphone information

 - Device type (smartphone, tablet, laptop, etc): smartphone
 - Device make (i.e. Samsung, OnePlus...): Xiaomi
 - Android version (i.e. Android 10): 13, 12
 - ONLY IF YOU'RE USING A CUSTOM ROM, SUCH AS LINEAGE OS:
   - ROM name (i.e. LineageOS, PixelExperience...):
   - ROM version:
   - ROM origin (official or unofficial?)
   - ONLY IF UNOFFICIAL ROM - LINK TO DOWNLOAD (XDA post, blog post, etc):
 - ONLY IF YOU'RE USING THE ORIGINAL OS INSTALLED BY THE MANUFACTURER:
   - ROM name (i.e. Samsung OneUI, Xiaomi MIUI): 
   - ROM version (i.e. OneUI 3.0):

## App information

- App version (find it in the About screen inside the app): All versions
- Installed from (Play Store, F-Droid, GitHub or self-built): All

## USB drive information

- USB drive capacity (i.e. 8 GB): 
- All information about the drive, AS SHOWN IN THE APP:

## USB port info

- Does the USB drive plug directly into your device without any adapters? (Yes/No): 

- If using a USB adapter, dock or hub:
  - USB port type ON THE DEVICE (put an "X" inside `[ ]`, like this: `[X]`):
    - [ ] USB type C
    - [ ] Micro USB
    - [ ] USB type A (those commonly found on non-Apple laptops and desktops)
  - USB port type ON THE USB DRIVE (put an "X" inside `[ ]`, like this: `[X]`):
    - [ ] USB type C
    - [ ] Micro USB
    - [ ] USB type A (those commonly found on non-Apple laptops and desktops)
  - Adapter make and/or where you bought it (Amazon, AliExpress, local store, bundled with phone...):
  - Adapter type (put an "X" inside `[ ]`, like this: `[X]`):
    - [ ] Compact, solid microUSB/Type-C to USB-A OTG adapter
    - [ ] Flexible microUSB/Type-C to USB-A adapter
    - [ ] USB-C dock with > 1 USB port (usually advertised for laptops)
    - [ ] Other type of USB hub with > 1 port

## ISO or image file information

- Image file format (ISO, DMG, IMG): 
- Image file size AS SHOWN BY THE APP: 
- Linux distribution name and version (if it applies, i.e. Ubuntu 20.04): 
- Is this image for a Raspberry Pi or similar? (Yes/No): 

## Additional info

Click "Write Apple® DMG image" >> "I UNDERSTAND" then throw this error.

lyk082401, 2023-09-08:
This is video link: [Screenrecorder-2023-09-09-05-26-45-25.mp4](https://cotools.github.io/gitnetdisk/Screenrecorder-2023-09-09-05-26-45-25.mp4)

depau, 2023-12-25:
Hi,
Thanks for reporting the issue. Unfortunately I decided to remove the DMG writing functionality in order to focus on improving stability and performance.

Sorry about that.

closed 2023-12-25 by depau