etchdroid/etchdroid #191: [Stupid text?]in TV Android4.4.4 run etchdroid

OYlili, 2023-10-02:
I am using version 1.3.32 of etchdroid. Although it can be installed and used, it crashes when selecting an iso file. I hope you can make it compatible with TVs or create a TV version when you have time, as it would be a bit silly but also fun to do so.

OYlili, 2023-10-03:
Can run etchdiord on TV
Although it can be run, it crashes when selecting "img/iso"
Is this due to a lack of external applications?
**(Because** previously, no file manager would also crash when selecting files)

thesreal, 2023-12-07:
Maybe your TV may not have a compatible USB port? Just saying, I might be wrong...

depau, 2025-05-02:
Hi! As you figured out it most likely crashes due to the lack of a file manager on Android TVs. I'm not planning to support running it on TVs.

closed 2025-05-02 by depau