etchdroid/etchdroid #198: [Bug] Write failed at block 0, 0.0 B written. Error: java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again

DINKIN, 2023-12-13:
## Description
Write failed at block 0, 0.0 B written. Error: java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again

## To reproduce
### Error message

```
Write failed at block 0, 0.0 B written. Error: java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again
```

### Device info

- Android version: 13 (API 33)
- Manufacturer: samsung (samsung)
- Model: SM-S908U1 (b0quew, hw name: qcom, board: taro)
- Build fingerprint: samsung/b0quew/b0q:13/TP1A.220624.014/S908U1UES3CWK1:user/release-keys

### App build info

- Version: 1.5.1 (13)
- Installed from: com.google.android.packageinstaller
- Build type: release (debug: false)

### Stack trace

```eu.depau.etchdroid.utils.exception.UsbWriteException: Write failed at block 0, 0.0 B written. Error: java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again
	at eu.depau.etchdroid.services.UsbApiWriteService.writeInputStream(UsbApiWriteService.kt:80)
	at eu.depau.etchdroid.services.UsbApiWriteService.writeImage(UsbApiWriteService.kt:111)
	at eu.depau.etchdroid.services.UsbWriteService.onHandleIntent(UsbWriteService.kt:41)
	at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:78)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:226)
	at android.os.Looper.loop(Looper.java:313)
	at android.os.HandlerThread.run(HandlerThread.java:67)
Caused by: java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again
	at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:181)
	at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.write(ScsiBlockDevice.kt:340)
	at eu.depau.etchdroid.services.UsbApiWriteService.writeInputStream(UsbApiWriteService.kt:78)
	... 7 more
```

depau, 2023-12-25:
Hi,
Does this issue still happen with the latest pre-release of the app?

DaveTmire85, 2024-11-29:
I am running the current deployment from the Play Store, and can also replicate this issue when attempting a write of an Ubuntu live iso to a USB.

depau, 2024-11-29:
This issue should be resolved in the latest releases.

@DaveTmire85 Changing the drive will most likely work. The Android system sometimes gets stuck and you need to reconnect the USB, restart the app or reboot the device.

I'll add a message suggesting a device restart since that should cover the vast majority of issues.