etchdroid/etchdroid #209: Unable to create ChromeOS installer

reejit, 2024-01-12:
The .bin file are available on the web and it is possible on balena etcher and gnome disks but not on etchdroid

depau, 2024-01-12:
Hi, can you please clarify the problem? Is EtchDroid failing? What version are you using?

reejit, 2024-01-13:
Yes, it's failing to write the .bin file. I'm using the latest version
available on fdroid

depau, 2024-01-13:
Unfortunately I can't telepathically connect to your phone to figure out what is going on :)

I wish I could, but I'm gonna need some help from you!

- Can you walk me through the steps to reproduce your issue?
- What error message do you get? Can you send a screenshot?
- What phone are you using?
- Are you capable of collecting logs via logcat?
  - If you aren't [I asked ChatGPT for you](https://chat.openai.com/share/56d40cd7-6309-4cac-98a1-0919bd1fc968), if you'd be able to do that and send me the data via email that'd be great! (don't attach it here, it may contain sensitive info)

reejit, 2024-01-14:
Currently I'm out of my country, I'll make sure when I get back I'll send
you the info

depau, 2025-05-02:
Hi! Closing this issue since I haven't heard back. See if it works with the most recent release. Let me know if you still have issues.