etchdroid/etchdroid #212: Bug report

rolexton, 2024-01-26:
### Error message

```
Write failed at block 294912, 151.0 MB written. Error: java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again
```

### Device info

- Android version: 13 (API 33)
- Manufacturer: POCO (Xiaomi)
- Model: M2102J20SG (vayu_global, hw name: qcom, board: vayu)
- Build fingerprint: POCO/vayu_global/vayu:13/TKQ1.221013.002/V14.0.3.0.TJUMIXM:user/release-keys

### App build info

- Version: 1.5.1 (13)
- Installed from: io.github.muntashirakon.AppManager
- Build type: release (debug: false)

### Stack trace

```eu.depau.etchdroid.utils.exception.UsbWriteException: Write failed at block 294912, 151.0 MB written. Error: java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again
	at eu.depau.etchdroid.services.UsbApiWriteService.writeInputStream(UsbApiWriteService.kt:80)
	at eu.depau.etchdroid.services.UsbApiWriteService.writeImage(UsbApiWriteService.kt:111)
	at eu.depau.etchdroid.services.UsbWriteService.onHandleIntent(UsbWriteService.kt:41)
	at android.app.IntentService$ServiceHandler.handleMessage(IntentService.java:78)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:211)
	at android.os.Looper.loop(Looper.java:300)
	at android.os.HandlerThread.run(HandlerThread.java:67)
Caused by: java.io.IOException: MAX_RECOVERY_ATTEMPTS Exceeded while trying to transfer command to device, please reattach device and try again
	at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.transferCommand(ScsiBlockDevice.kt:181)
	at com.github.mjdev.libaums.driver.scsi.ScsiBlockDevice.write(ScsiBlockDevice.kt:340)
	at eu.depau.etchdroid.services.UsbApiWriteService.writeInputStream(UsbApiWriteService.kt:78)
	... 7 more
```

depau, 2024-02-07:
Hi! Can you please try with the latest release candidate? You can download it from F-Droid or join the open beta on Google Play: https://play.google.com/apps/testing/eu.depau.etchdroid

rolexton, 2024-02-09:
[image]

depau, 2024-02-09:
Hi, you need to wait some time after joining the testing channel before it becomes available. You can also get it from here: https://github.com/EtchDroid/EtchDroid/releases/tag/v1.9.rc1

closed 2025-05-02 by depau

PeachyBird, 2026-06-10:
Doing the same thing to me on a Google Pixel 6a