etchdroid/etchdroid #224: Internal SD

webmaster-exit-1, 2024-05-06:
Was stopping by the repo to comment on how well this Android application works, then I noticed "_Not working: Internal SD card_".
And it got me thinking... 

Have you tried using the `sm` command to make the internal SD card world-readable and writable?

- The `sm` command is part of the Android Storage Manager, and you can use it to change the permissions and mount options for storage devices.

**Example**:

```bash
adb shell
sm set-virtual-disk:enabled true
sm set-permission /sdcard write
sm set-permission /sdcard read
```

This **WILL** enable virtual disk support and set the permissions for the internal SD card to be writable and readable by all users. I'm almost 100% sure you **WILL NOT** be able to boot into the newly flashed internal SD card. But..., I think this **WILL** make it flash-able.

> If it works; cool. If not; meh.

**EtchDroid**
Great Android App. Love it.
Keep up the good work.
**HAPPY HACKING**

depau, 2024-05-06:
Hi,
That method is out of scope since the app aims to work exclusively using regular Android API permissions, and what you suggest requires ADB access.

Besides, the app needs raw block device access to the SD card, not filesystem access. That's absolutely impossible on Android on any built-in device peripheral and it's currently only possible (not without hacks) on USB devices.

closed 2024-05-06 by depau

webmaster-exit-1, 2024-05-14:
I did not know that. For sure man. Love EtchDroid non the less, great app. Take care.