etchdroid/etchdroid #226: Etchdroid "Cannot read USB device"?

Justin2916, 2024-05-16:
I am having some trouble trying to use Etchdroid to flash the latest image of Retropie for the Raspberry Pi onto a SD card.

I am on a Samsung Galaxy Note 9(SM-N960U1) running Android 10. deivce is fully updated as is Etchdroid.

The SD card is a 256gb Samsung Evo formatted using Androids built in Format option. The SD card reader is a off-brand "uni" that has dual USB(USB-C & standard type A) as well as SD ports(one Micro SD port and one larger SD port)

Etchdroid opens fine and lets me choose the Retropie Image but after I select the image and usb device it goes to a black screen for several minutes before going to a "ready to write" page and at the bottom says "Cannon read USB device" and when i hit the "write" button on the bottom right corner it says "Cannot write image to USB  drive"
[image]
[image]
[image]

The first photo just shows the USB device. The second Photo shows the app asking permission to acces the SD card reader(i select yes). and the last photo shows the screen and "Cannot Read USB device" that pops up after the black screen.

Justin2916, 2024-05-16:
Edit:
I figured out it was my SD reader after trying a couple different apps. Even though i was only trying to Flash one SD card, the apps wouldnt recognize the reader correctly until i had both SD slots full

depau, 2025-05-02:
Hi! Raspberry Pi images work but you need to extract them. Thanks for reporting back, and sorry for not replying earlier. Best regards

closed 2025-05-02 by depau